FreeCAD/FreeCAD #20586: Create a Fillet one way- it works, but create a fillet another way it fails.

JDMils, 2025-04-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Here is the post as reference:
https://forum.freecad.org/viewtopic.php?t=95965&start=11

I created a fillet on a model using a specific set of steps, and the fillet fails. I then created the same fillet using the actions in a different order, and the fillet works. And this is consistent & repeatable.

News flash- depending on which order you execute the fillet, it will either ALWAYS crash or NEVER crash the model!

I tried this:
* With just the text block selected "A_Block" and not the model on screen, click Part Fillet.
* Adjust the Radius to 0.01mm.
* Warning- no edges selected. Click radio button Fillet Parameter | Selection | Select Faces then select the top face.
* Click OK.
==> Model crashes.

Now change the step order around and it works:
* With just the text block selected, "A_Block" and not the model on screen, click Part Fillet.
* Click radio button Fillet Parameter | Selection | Select Faces then select the top face.
* Adjust the Radius to 0.01mm.
* Click OK.

Fillet now works!!?? And if I use the 2nd method, no matter how many times, it will always work! So now no need to create incremental fillets to get to the size I want!

This is the video of me trying both methods:
https://youtu.be/ZFtB_wBT0iw

[testTexDraft-001.zip](https://github.com/user-attachments/files/19562529/testTexDraft-001.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/106.57/1.5
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.18
  * Curves 0.6.58
  * fasteners 0.5.38
  * Gridfinity 0.11.4
  * InventorLoader 1.5.1
  * lattice2 1.0.0
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jonzirk76, 2025-04-03:
I can confirm this behavior on the test file provided, however I was not able to replicate this on a sketch I extruded myself.  Maybe it has something to do with the cut.

JDMils, 2025-04-03:
As you know, fillets are flakey in FreeCAD (I should get that on a bumper sticker!). You have to get that type of angle in your model where it will have issues. Of course if you try to recreate it, you may not get that "angle" right enough for the code to choke on.

maxwxyz, 2025-04-04:
@pieterhijma FYI

maxwxyz, 2025-11-08:
I can manually assign the label but with the stale action you would get a reminder to check if it might be solved

stefano2734, 2026-09-17:
I see here only last test with 2 years old occt 7.8.1.
https://github.com/Open-Cascade-SAS/OCCT/releases/tag/V7_8_1
Occt 7.9.3 and 8.0.1 are here actual available with some solved issues also to FreeCad. 
They are available with FreeCad Libpack 3.4.0 and 3.5.5.
Improvement to fillets will come with PRs in 8.0.2 and 8.1.0.

pieterhijma, 2026-09-29:
This is not an OCCT issue, but a UX issue that causes some confusion. 

The unselected edges have a default radius of 1 mm. When first providing the radius input without having edges selected, there is a warning that the value is not transferred to any edge. When users then select the top face, the radius is not updated with the value that is provided as input, but it keeps the radius that it had before. I think this is a conscious decision to ensure that users can choose different radii for different edges. The warning actually tries to explain this.

The fillet is correctly applied by OCCT. A fillet of 1.0 mm is simply a fillet that does not work and is reported by OCCT as such.

I recommend to close this issue.