FreeCAD/FreeCAD #20590: Lost update menu in Preferences for AppImage

tarman3, 2025-04-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In weekly build 40971 and 41127 lost menu for update AppImage

### Full version info

```shell
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41127 (Git) Conda AppImage
Build date: 2025/04/01 15:11:50
Build type: Release
Branch: main
Hash: db08735e106561f3a11cf08e02fbf5d9e8fbf6b2
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/118.293/1
Installed mods: CurvedShapes 1.0.13; lattice2 1.0.0; Silk 0.1.5; freecad.gears 1.3.0; btl 0.9.9 (Disabled); fasteners 0.5.38; MeshRemodel 1.10.37; Curves 0.6.60; sheetmetal 0.7.22; Defeaturing 1.2.3
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-04-02:
This was asked in the forum see https://forum.freecad.org/viewtopic.php?p=819319#p819319

FEA-eng, 2025-10-04:
@tarman3 Is this issue still relevant?

closed 2026-01-11 by tarman3