FreeCAD/FreeCAD #20595: `Create a dantum` is missing in the prerelease

code-forge-temple, 2025-04-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

in the weekly prerelease build 

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/109.57/1
Installed mods: 
  * Curves 0.6.60
  * freecad.gears 1.3.0

i cannot find any more the `create a dantum` option.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/109.57/1
Installed mods: 
  * Curves 0.6.60
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-02:
It is now in the toolbar where you find create part

closed 2025-04-02 by maxwxyz

code-forge-temple, 2025-04-02:
> It is now in the toolbar where you find create part

@maxwxyz i have downloaded today (before i've opened the issue) the latest FreeCAD 1.1.0dev and I cannot find it neither in the `Part Design` menu nor in toolbars (in version 1.0.0 it is present in both toolbars + the Part Design menu section):

[image]
[image]