FreeCAD/FreeCAD #20599: UI: There is no visual reference to the Sketch that is being edited.

leoheck, 2025-04-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Right now, when a Sketch is being edited, I cannot tell which Sketch is being edited.

There is no visual indication of which Sketch is being edited when looking at the Objects Tree and the Task panel. See an example:

[image]

The Sketch should be highlighted in the tree, and its Name (aka Label) could be displayed on the Task panel. 

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-04-04:
The object being editing colour should be clear:

[image]

but the sketch label can be added to the Solver Messages reasonably easily,:

[image]

[image]

leoheck, 2025-04-07:
Yeah, @maxwxyz you are right. Using the default dark theme, I can see the selected sketch.

The color theme is not good (did not restart freecad after changing themes thogh), no matter if Sketch starts as visible or not, but this is something that can be improved. 

[image]

[image]

leoheck, 2025-04-10:
The PD Feature being edited could also be placed in the Task. 

[image]

leoheck, 2025-05-06:
Any plans to have this improved?

leoheck, 2025-06-19:
This does not impact the Sketches edition only. It does for Features and maybe other objects, too.

Other important Features are the Part Design tools, which are always buggy on complex designs, or in designs that have the `Skip Recompute` enabled. Most of the time, the Transform tools. For instance, in this latest weekly `42180`, multiple Features remain visible after closing the one being edited. And this kind of messes up the whole design process, its crazy.

Can we think of other ways of working around this for a while?
1. Adding an icon in the tree view, that shows what the object is in edit mode... maybe a pencil or something.
1.  Adding the Label of the Object to the Task panel.
1. Printing a message to the Report view.

#1 and #2 can be done together, and they can remain after solving this issue with the missing highlight of the object being edited.

So the highlight missing, which is an important feature, won't impact much on the design process.

MisterMakerNL, 2025-06-19:
It's only missing if you use opentheme with the item background, the Freecad stylesheet use the overlay effect. 
I don't think the current solution is all that bad.

leoheck, 2025-06-19:
I do use OpenTheme as I think many others do. But I don't use the overlay effect, and most of the time cannot say which object I am editing. This, of course when I am updating existing objects. When I am creating a new one, this is not a problem, et al. 

What is the current solution, by the way? I don't see any solution.

MisterMakerNL, 2025-06-19:
Don't use Opentheme until someone fixed it.

leoheck, 2025-06-19:
haha, I see. Well the original dark theme sucks.

MisterMakerNL, 2025-06-19:
There are more themes, also how old are you?