FreeCAD/FreeCAD #20601: Color of selected line in sketch is not as set in preferences

resc-git, 2025-04-02:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In Preferences/General/Selection there are two options to define the color of selection and preselection.
While these settings take effect as expected in Part Design, this is not the case in Sketcher.
When you move the mouse over a line, the color of the Preselection is applied as expected. However, when selecting the active object, the color does not immediately change to the color defined in Preferences/General/Selection. It changes to a different color until the mouse pointer is moved away. 
My expectation would be that it would use the color from the settings even when the mouse pointer is on the line.
In all other views it behaves as I expect.
(With default color settings it is hard to distinguish selected from unselected lines/objects)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * A2plus 0.4.65
  * ExplodedAssembly
  * freecad.gears 1.2.0
  * ThreadProfile 1.92.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jonzirk76, 2025-04-03:
This behavior happens on my system as well.

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

```

FEA-eng, 2025-12-14:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251209 (Git shallow)
Build date: 2025/12/09 13:53:17
Build type: Release
Branch: grafted,grafted
Hash: 5ad6a1ba58c665c23f1d217c23e74379d8c1a5a4
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

kaiwas, 2025-12-15:
> if this is still the case, yes please.

Halfway up.
The color can be adjusted in the settings.
But the vertex size cannot.

[image]

I created a macro that toggles the vertex size in sketches.

https://github.com/user-attachments/assets/0a5d5ce3-ff55-4258-b5ca-8ba30ef3ec15

```
import FreeCAD

def set_sketch_vertex_size():
    doc = FreeCAD.ActiveDocument
    

    if doc is None:
        FreeCAD.Console.PrintMessage("No active document found.\n")
        return
    

    for obj in doc.Objects:
        # Проверяем объекты. Условие "Если объект скетч".
        if obj.TypeId == 'Sketcher::SketchObject':
            if obj.ViewObject.PointSize <= 4: # Проверяем размер вершин. условие "Если меньше или равно 4"
                obj.ViewObject.PointSize = 11  # Если да то ставим размер вершин 11 пикселей
            else:
                obj.ViewObject.PointSize = 4 # Иначе ставим размер вершин 11 пикселей

# Запуск функции
set_sketch_vertex_size()
```

ICO

[image]

ErtugrulEge, 2026-02-22:
> With default color settings it is hard to distinguish selected from unselected lines/objects

I agree. I  use FC Light them and in Sketcher lines are almost blac and selectin color also almost black as default. Should I open a new issue?

tigert, 2026-03-04:
> Just today I found out that there is no global setting responsible for the size of the vertices and the thickness of the sketch lines outside the editing mode. Should I create a separate issue about this?

This (indeed offtopic here) issue seems to work now:

[image]

Preferences > Part Design > Shape Appearance > Vertex size also is used for sketches outside Sketcher.

ErtugrulEge, 2026-10-05:
Seems still relevant 

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 597.06/NVIDIA Corporation/Quadro RTX 4000/PCIe/SSE2
Installed mods:
  * Curved Shapes 1.0.15
  * Curves workbench 0.6.81
```