FreeCAD/FreeCAD #20607: Inconsistent class name and Type

aiksiongkoh, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In the class definition below, "Type" should be "Fem::MeshGmsh" for consistency with other classes in FEM workbench.

...\src\Mod\Fem\femobjects\mesh_gmsh.py
class MeshGmsh(base_fempythonobject.BaseFemPythonObject):
    """
    A Fem::FemMeshObject python type, add Gmsh specific properties
    """

    Type = "Fem::FemMeshGmsh"

### Full version info

```shell
This is in source code of FreeCAD main repo as of 2025-04-01.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-04-03:
cc @marioalexis84

marioalexis84, 2025-04-03:
In Fem, the Mesh object has the prefix 'Fem' in its class name, which is somewhat redundant since it's already in the Fem namespace. However, I guess the name was given to differentiate it from the mesh in the Mesh module.
It makes sense to me to use Fem::FemMeshGmsh and Fem::FemMeshNetgen, as these classes are Python proxies for Fem::FemMeshObjectPython objects.

luzpaz, 2025-04-08:
@marioalexis84 verdict ?