FreeCAD/FreeCAD #20608: SVG files exported from GIMP cannot be read

osamuhimura, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If the language setting in GIMP is set to 2-byte characters, such as Japanese, the name used for the path id in the SVG file will automatically use 2-byte characters when the path is extracted, and FreeCAD will fail to load the file with an error.

It is possible to work around this by changing the name used when extracting paths in GIMP to an English name.
However, Autodesk Fusion and other software can import even 2-byte characters, so we report this as a problem.

[https://x.com/HimuraMechatro/status/1906720913808404493](url)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Japanese/Japan (ja_JP)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.12
  * fasteners 0.5.37
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-04-03:
@osamuhimura thanks for the bug report. 
any chance you can:  
* zip the example file and drag/drop it into the github textbox so it attaches to this ticket and we can then test it?  
* could you, as asked by Roy-043 above, install the development version[^1] and test to see if this issue still persists ?

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

osamuhimura, 2025-04-04:
Thank you for your reply.
I have uploaded the SVG file which I have checked here.
I will look into installing the latest Development version of FreeCAD, as I don't know how to do that yet.

[Test_svg.zip](https://github.com/user-attachments/files/19597023/Test_svg.zip)

osamuhimura, 2025-04-04:
I checked with the development version and was able to open the problematic SVG file.

[image]

maxwxyz, 2025-04-04:
I'll close this issue if already fixed in the dev version.

closed 2025-04-04 by maxwxyz

Syres916, 2025-04-07:
Having looked at the 1.0 branch it appears as though the fixes haven't been backported so I'll raise a backport PR later today/tomorrow.