FreeCAD/FreeCAD #20610: Integrated Assembly, joints not recomputing when liked to variables objects (properties,varset, dynamic data)

CharlesMir, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Some constraints associated to property like objects (properties, varset, dynamic data) does not recompute correctly automatically, a manual recompute needs to be done for the changes to be taken into account.

Here are the steps to recreate the problem:

Create an assembly project
Create a varset
In the varset create a "Length1" variable (distance type)
On the part workbench, create two cubes
In the varset create a "Length2" variable (distance type)
Ground one cube
Assign Length1 between two faces of the cubes
Assembly should recompute
Assign Length2 between two faces of the cubes
Assembly should recompute
Modify Length1, assembly should recompute accordingly
Modify Length2, assembly does not recompute unless Joints "Distance001" is manually recomputed (right click, recompute object)

I tried also with properties and dynamic data objects, and experienced the same behavior. 
Link to freecad post : 
[https://forum.freecad.org/viewtopic.php?t=95974](url)

[AssemblyVarset.FCStd.zip](https://github.com/user-attachments/files/19580741/AssemblyVarset.FCStd.zip)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39841 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7a5a3d1ffcd6a5a01c90f6d17f5dd7eb8f6fa7aa
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods:
* Curves 0.6.60
* lattice2 1.0.0
* ThreadProfile 1.93.0
* fasteners 0.5.37
* DynamicData 2.73.0
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-07:
@philjynx is it a dupe ?

philjynx, 2025-05-07:
@luzpaz it looks like the same issue to me

closed 2025-05-13 by luzpaz