FreeCAD/FreeCAD #20613: Dimension overlapping arrows

mpfusion, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Vertical dimensions overlap the arrows if the dimension line is shorter than the dimension:

[image]

### Expected behaviour
Enough clearance so the dimension and arrows stay clear of the dimension in all cases.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/lightdm-xsession/xcb)
Architecture: x86_64
Version: 1.1.0dev.41079 (Git)
Build date: 2025/03/30 22:14:19
Build type: Release
Branch: main
Hash: 487c62a663a567a08d87360ed6c74dcbc998081f
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0197/1
Installed mods: 
  * fasteners 0.5.37
  * kicadStepUpMod 11.5.7
```

### Subproject(s) affected?

TechDraw

### Anything else?

Example project:
[overlap.tar.gz](https://github.com/user-attachments/files/19585174/overlap.tar.gz)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jonzirk76, 2025-04-06:
There doesn't seem to be that overlap in the file provided when I open it.

[image]

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

```

benj5378, 2025-04-07:
Spacing seem different between horizontal and vertical

WandererFan, 2025-05-21:
Is this still an issue?  I can't reproduce. Is there an unusual font involved?

[image]

```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41471 +1 (Git)
Build date: 2025/05/09 19:58:15
Build type: Unknown
Branch: PR21169
Hash: fc464814798bcd03fea9d4bca10cd181b8f85171
Python 3.10.12, Qt 5.15.2, Coin 4.0.0, Vtk 7.1.1, boost 1_74, Eigen3 3.4.0, PySide 5.15.2
shiboken 5.15.2, SMESH 7.7.1.0, xerces-c 3.2.3, OCC 7.8.0
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 115/141.585/1
Installed mods: 
  * fasteners 0.5.21
```

WandererFan, 2025-05-21:
[image]

mpfusion, 2025-05-21:
Yes, still an issue on `FreeCAD 1.1.0 Revision: 41736 (Git)`:

[image]

I'm not aware that I changed the font. *Dimension Format* reads: `Font: osifont`

benj5378, 2025-05-22:
@WandererFan looking at this again, this might very well be related to #21169. @mpfusion could you try out the PR?

mpfusion, 2025-05-22:
Merging https://github.com/FreeCAD/FreeCAD/pull/21169 indeed appears to fix this issue:

[image]

I didn't run any futher tests, though.