FreeCAD/FreeCAD #20614: Ubuntu 20.04, Freecad 1.0, crashes on interaction with assembly

nameofuser1, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I face very frequent crashes for no obvious reason. Usually happens when I try to interact with an object while assembly is active. Here are some logs of two sequential crashes:

`Ball_Bearing_4_7_2_5#Joint002 is unconnected to a grounded part so it is ignored.
244.127 <App> Document.cpp(2903): Ball_Bearing_4_7_2_5#Joint002 still touched after recompute
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f479e61a210]
#1  0x7f479dfaa4a5 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0x25
#2  0x7f479e161b9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0xea
#3  0x7f479e1b1348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0x58
#4  0x7f479e1cc13a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0x2a
#5  0x7f479dfaa4e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0x68
#6  0x7f479e161b9a in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0xea
#7  0x7f479e1b1348 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0x58
#8  0x7f479e1cc13a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0x2a
#9  0x7f479dfaa4e8 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0x68
#10  0x7f479dfb426c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0xac
#11  0x7f479dfab1ba in SoAction::apply(SoNode*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0x2ea
#12  0x7f479e17dad2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libCoin.so.80+0x72
#13  0x7f47a14f414e in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libFreeCADGui.so+0x1e
#14  0x7f47a14ff618 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libFreeCADGui.so+0x58
#15  0x7f47a1509360 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libFreeCADGui.so+0x40
#16  0x7f47a1511bc6 in Gui::TinkerCADNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libFreeCADGui.so+0x2f6
#17  0x7f47a150b265 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libFreeCADGui.so+0x65
#18  0x7f47a1534c3b in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libFreeCADGui.so+0x6b
#19  0x7f47a14ea9e5 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libFreeCADGui.so+0x1a5
#20  0x7f479ecbc8bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Core.so.5+0x7b
#21  0x7f479f7e939e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#22  0x7f479f7f08f9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Widgets.so.5+0xa39
#23  0x7f47a1126559 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libFreeCADGui.so+0xc9
#24  0x7f479ecbcb28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Core.so.5+0x108
#25  0x7f479f7ef3eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#26  /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7f479f83fe6f]
#27  /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7f479f8430c9]
#28  0x7f479f7e93ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#29  0x7f47a1126559 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libFreeCADGui.so+0xc9
#30  0x7f479ecbcb28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Core.so.5+0x108
#31  0x7f479f14b3ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Gui.so.5+0x84c
#32  0x7f479f1210bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Gui.so.5+0xac
#33  /tmp/.mount_FreeCAmfEjJD/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7f4798373bea]
#34  /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7f479ac09ad7]
#35  /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7f479ac0cac7]
#36  /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7f479ac0d0ce]
#37  0x7f479ed10a36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Core.so.5+0x66
#38  0x7f479ecbb70b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Core.so.5+0x11b
#39  0x7f479ecc3212 in QCoreApplication::exec() from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libQt5Core.so.5+0x82
#40  0x7f47a1085744 in Gui::Application::runApplication() from /tmp/.mount_FreeCAmfEjJD/usr/bin/../lib/libFreeCADGui.so+0x8a4
#41  /tmp/.mount_FreeCAmfEjJD/usr/bin/freecad(+0x78b7) [0x561a942868b7]
#42  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f479e5fb0b3]
#43  /tmp/.mount_FreeCAmfEjJD/usr/bin/freecad(+0x7c2a) [0x561a94286c2a]
`

`
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f59565c0210]
#1  0x7f58b6cdbbae in MbD::PartFrame::removeRedundantConstraints(std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> > >) from /tmp/.mount_FreeCAnoeeba/usr/lib/./libOndselSolver.so.1+0x1e
#2  0x7f58b6cce137 in MbD::Part::removeRedundantConstraints(std::shared_ptr<std::vector<unsigned long, std::allocator<unsigned long> > >) from /tmp/.mount_FreeCAnoeeba/usr/lib/./libOndselSolver.so.1+0x47
#3  /tmp/.mount_FreeCAnoeeba/usr/lib/./libOndselSolver.so.1(+0x3154cc) [0x7f58b6cee4cc]
#4  0x7f58b6d32a8a in std::function<void (std::shared_ptr<MbD::Item>)> std::for_each<__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)> >(__gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, __gnu_cxx::__normal_iterator<std::shared_ptr<MbD::Part>*, std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >, std::function<void (std::shared_ptr<MbD::Item>)>) from /tmp/.mount_FreeCAnoeeba/usr/lib/./libOndselSolver.so.1+0x5a
#5  0x7f58b6d336b6 in MbD::System::partsJointsMotionsLimitsDo(std::function<void (std::shared_ptr<MbD::Item>)> const&) from /tmp/.mount_FreeCAnoeeba/usr/lib/./libOndselSolver.so.1+0x76
#6  /tmp/.mount_FreeCAnoeeba/usr/lib/./libOndselSolver.so.1(+0x129f74) [0x7f58b6b02f74]
#7  0x7f58b6d34000 in MbD::System::runPreDrag(std::shared_ptr<MbD::System>) from /tmp/.mount_FreeCAnoeeba/usr/lib/./libOndselSolver.so.1+0x110
#8  0x7f58b6b4bc02 in MbD::ASMTAssembly::runPreDrag() from /tmp/.mount_FreeCAnoeeba/usr/lib/./libOndselSolver.so.1+0x1f2
#9  0x7f58bc221523 in Assembly::AssemblyObject::solve(bool, bool) from /tmp/.mount_FreeCAnoeeba/usr/lib/AssemblyApp.so+0x223
#10  0x7f58bc221806 in Assembly::AssemblyObject::preDrag(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> >) from /tmp/.mount_FreeCAnoeeba/usr/lib/AssemblyApp.so+0x36
[... 47 lines cut ...]
#58  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f59565a10b3]
[... cut ...]
### Full version info

```shell
OS: Ubuntu 20.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-15:
Please provide an example fiile and steps to reproduce. Could you test the issue on the weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

nameofuser1, 2025-04-16:
@maxwxyz unfortunately I can't. Shortly after I created the issue the new crash corrupted the files and freecad can't restore them

luzpaz, 2025-05-08:
> unfortunately I can't. Shortly after I created the issue the new crash corrupted the files and freecad can't restore them

Oooof...looks like #18044 has struck again

PaddleStroke, 2025-07-17:
On my side (windows 11) I get the following : 
- Activate the assembly
- Try to click and drag the FrameVerticalProfile002 -> crash

and : 
- Activate the assembly
- Solve assembly -> crash

Then I tried to reproduce the steps as in your video. During joint creation I get : 
```
17:18:39  0 <Assembly> AssemblyObject.cpp(160): Solve failed: invalid vector subscript
17:18:39  0.0744573 <Assembly> AssemblyObject.cpp(160): Solve failed: invalid vector subscript
17:18:39  0.147692 <Assembly> AssemblyObject.cpp(160): Solve failed: invalid vector subscript
```

But no crash. Until I try to click and drag.

Reviewing your video now, and I think that on your video it may also crash at the same time.

PaddleStroke, 2025-07-17:
[image]

Ok so now I have a simplified version of the file that is crashing.

There is a distance joint that is redundant. Which makes me think that MbD::PartFrame::removeRedundantConstraints is indeed crashing.

PaddleStroke, 2025-07-17:
Ok found it. It crashes because of the 'fixed joint bundling' feature which is bundling the fixed parts together for the solver to be faster during drag.
The issue is that after this bundling, the conflicting distance joint point to the same part on both ends. Which is probably not handled correctly in the solver and end up in a crash.

So it should just need to ignore joints that point on both end to the same mbd part. Let's see

closed 2025-07-18 by chennes