FreeCAD/FreeCAD #20615: [UI/UX] Add a method for selecting entire objects to the selection filter.

kaiwas, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, the 3D window implies the selection of some components of the object. But in reality, there is always a double selection:
1) The entire object (highlighted in the view tree)
2) An element of the object (Vertex/edge/face).

When you hover the mouse, a constant analysis is performed to select a sub-object. This is not a problem if the object is small. But if the model is very complex, such an analysis takes quite a long time.

https://github.com/user-attachments/assets/5d0e9ab5-be5e-417c-8c40-bbf3ffd8b2da

I suggest adding "Object" to the selection filter. In case of a click, the entire object will be highlighted, as if we had clicked on the object in the view tree. An approximate concept is in the picture.

[image]

In addition, I can say that "double selection" is a bit unusual for new users. It does not immediately become clear that an object is being selected, despite the fact that only one face of the object is highlighted.
Not all actions require selecting any of the sub-elements of the object; it is enough to select the entire object.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41127 (Git) Conda AppImage
Build date: 2025/04/01 15:11:50
Build type: Release
Branch: main
Hash: db08735e106561f3a11cf08e02fbf5d9e8fbf6b2
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Test
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.60
  * freecad.gears 1.3.0
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-04-03:
Let's imagine that a Boolean operation needs to be performed.
I select objects in the 3D window (faces and edges are selected). Then, for some reason, I rotate the 3D view.
It becomes completely unclear which objects are already selected.

[image]

Yes, I can do this using tree view, but it is not always convenient.

maxwxyz, 2025-04-04:
@FreeCAD/design-working-group any feedback on this?

pierreporte, 2025-04-04:
It’s a great proposal. The problem is, what is an object? For parts containing multiple bodies, should it select bodies or parts? The example shown are bodies (from Part workbench). How would it work for BIM?

kaiwas, 2025-04-04:
> How would it work for BIM?

Works in a completely similar way. On the screen, "subelements" of objects are selected. Their names are highlighted in the tree. 
It would be convenient to work at the level of the object as a whole and only when it is necessary to switch to the level of vertices/edges/faces

https://github.com/user-attachments/assets/1df58d99-5fb4-47f3-bbe9-ab5af917349c

maxwxyz, 2026-08-17:
@tritao FYI (for object selection filter = default in the current drag to select mode). Then we could get rid of the extra box element select tool and make the standard drag to select behavior respect selection filters.