FreeCAD/FreeCAD #20617: BIM: Window frame property: terminology misconception

ErtugrulEge, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hello, BIM window has a frame property but it is not clear what it means.Now the "frame" property controls the frame depth. I think we need two property.                     
1.Frame width
2.Frame depth

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-05-20:
@marcuspollio @maxwxyz @ErtugrulEge I stumbled upon this today while attempting to write a docstring for the `Arch.makeWindow` function. I think this issue is valid. Please consider reopening it.

The terminology is indeed confusing. Some findings:

1. The `Frame` property defines the thickness of the outer frame. It is named `Frame` (as per the property, this cannot be changed) in the Property Editor. It is named `Thickness` in the Task panel when editing the window. That measurement corresponds to the "frame depth" measurement in the issue description screenshot.
2. The "frame width" in the screenshot is defined by the distance between the two concentrical rectangles defined as wires in the base sketch and put together as a component in the Task panel's window editor. While this might not be intuitive, I don't think there is much to do here.

For point 1, I would propose either:

A. Renaming the `Frame` property to `FrameThickness`. I feel this would be the cleanest approach, but renaming properties breaks backwards compatibility.
B. Renaming the `Thickness` label to `Frame thickness` in the window editor. Minimal change, clear, but does not match the property name. There is the tooltip for that, though.
C. Renaming the `Thickness` label to `Frame` in the window editor.  Minimal change, not entirely clear, but does match the property name. The tooltip can be modified to explain it's a thickness.

I understand the refactor is happening, but given that the change would be minimal, and that there is no set date for the refactor to land, or whether it will make to 1.1, it might be worth tackling this. Creating windows is difficult enough as it is, so removing a source of confusion can only be beneficial to users.

I'm considering to submit a PR that implements proposal B. Thoughts?

ErtugrulEge, 2025-05-20:
According to me, option A is the solution. `Frame` property is in fact `FrameThickness` property and also  we can add another "FrameWidth" property to windows and doors. `Frame`  doesn't mean anything on its own but I don't know about development stuff. I also want to hear other ideas @kaiwas @semhustej

semhustej, 2025-05-20:
I am not a native English speaker, so I might be completely wrong, but according to my quick research:

[image]

**W1 is Frame Depth
H1 is Frame Thickness**

Sources: 
This is window panel, but the depth-thickness orientation should be the same for window frame: https://ifc43-docs.standards.buildingsmart.org/IFC/RELEASE/IFC4x3/HTML/lexical/Pset_WindowPanelProperties.htm
https://community.osarch.org/discussion/2777/designing-a-new-window-door-generator/p1

`
A. Renaming the Frame property to FrameThickness. I feel this would be the cleanest approach, but renaming properties breaks backwards compatibility.`
As pointed above, this should be FrameDepth. I don't think breaking backwards compatibility or creating significant changes is a good idea if Yorik is currently working on refactoring the tool.

`
B. Renaming the Thickness label to Frame thickness in the window editor. Minimal change, clear, but does not match the property name. There is the tooltip for that, though.`
I don't think that would be right as we are not talking about Thickness but about Depth..

`C. Renaming the Thickness label to Frame in the window editor. Minimal change, not entirely clear, but does match the property name. The tooltip can be modified to explain it's a thickness.`
I would not use `Frame` for the label, as the `Frame` parameter is different than the number that user inputs to this input filed.

**My conclusion:**
From the 'research' above, the semantically correct  change would be to rename the `Thickness` label in window editor to `Frame depth`.

furgo16, 2025-05-20:
That sounds good, thanks. I'll submit a PR with option B, but with Frame depth. I would add to that that the tooltip for the `Frame` property should be improved to mention depth too.

> According to me, option A is the solution. 

That makes sense, but whenever we rename a property, in a FreeCAD release, previous releases will not be able to open the file, as they would expect the old name. That's why I prefer the other approach.

kaiwas, 2025-05-20:
> other ideas [@kaiwas](https://github.com/kaiwas)

As for me, I don't understand the current situation. It's so unclear that I try not to think about it )))
It's enough for me that I can choose the "type" of the window (1, 2 or 3 frames). I don't need anything else for now.
But the terms that differ at the time of creation and in the properties panel are bad.

furgo16, 2025-05-20:
Note: the issue we're talking about is not in the `Window options` dialog (the one that appears when you click on the `Arch_Window` command), but the `Window elements` dialog that appears when double-clicking an existing window on the tree view to edit it. To see the actual text discussed here, you'll need to select a component in that dialog and edit it.

closed 2025-06-02 by yorikvanhavre