FreeCAD/FreeCAD #20618: UX: Clean equation fields automatically when there is only a single numeric value.

leoheck, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Sometimes, we automatically add single values to the Equations fields due to muscle memory or quick edits. 

So, if the Equation has a single numeric (integer/float) value, it could be automatically downgraded to a simple value instead of an equation. 

It will be like a clean up.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-04-03:
For the sake of illustration, this would work on cases like this

[image]

FlachyJoe, 2025-04-03:
I disagree : expression allow to define a more precise value than the property editor. If you convert the expression to a value you lose this.

[image]

leoheck, 2025-04-03:
Hm, yeah, considering this use case, this seems right. But now I am wondering if the user puts `10.0001` in the field, it should be used as the user put, isn't it?

FlachyJoe, 2025-04-03:
Displayed max decimal vs Used max decimal. 
When you type a more precise value than the max decimal preference in the property editor, the typed value could be kept and used but displayed truncated and should come back during the edition. I'm not sure it's feasible.

leoheck, 2025-04-03:
Yeah, right. I thought it was already like that.

FlachyJoe, 2025-04-03:
More: you could argue a static expression should be converted to its value but it will be even worse

[image]

I really type `10/3` in the property editor, the value is computed but set at the defined precision. If you use such value e.g. to distribute parts on a length you'll quickly have problems filling the entire path.

leoheck, 2025-04-03:
The static expression is nice to have as it is since it keeps the terms explicit.

kaiwas, 2025-04-04:
> but displayed truncated

By the way, this point is far from obvious. In my opinion, for values ​​whose precision is beyond the display limits, there should be some kind of warning sign.
At the moment, the precision may not correspond to the set setting (for example, I have 1 digit after the point). And I have no way to find out about this.

Maybe for the value 3.33333 with number decimals = 1, 3.3**~** should be displayed. This will make it clear that the precision is beyond the setting. At the same time, all the signs should be displayed when editing.

Watch the video. It feels like I'm losing 0.0333 somewhere. Although the visible result is always the same.

https://github.com/user-attachments/assets/dc67ef17-f1da-4589-bf18-48320a5e161b

Maybe we should create a discussion to discuss this issue? It seems quite important to me.

The test file is attached. (remove the zip extension)

[Check.FCStd.zip](https://github.com/user-attachments/files/19602527/Check.FCStd.zip)

maxwxyz, 2025-04-04:
@FreeCAD/design-working-group what is your opinion on this?

kaiwas, 2025-04-04:
The issue discussed here could be partially resolved if there was a faster way to "clean" expressions for several properties at once. This would simply simplify the issue of value processing (we can call it a semi-automatic mode)
https://github.com/FreeCAD/FreeCAD/issues/14605

pierreporte, 2025-04-04:
I don’t have a strong opinion either way, but I would lean towards being against such change, because it could be unexpected.

kaiwas, 2025-04-04:
Maybe it could be some kind of sign at the end of the field. When hovered over, it would indicate that the displayed value is different from the calculated one.

[image]

leoheck, 2025-04-04:
This last example makes sense.

If we can have (resulting) values with more than the max allowed decimal places, then it makes sense to have a sign to indicate this. The tooltip is a nice way to display it, but it should also have another visible sign so the user will know it has to be over the mouse there since what is being shown is half the truth. Ah yeah. In the above image, it is this asterisk at the end of the field.

Now, something different is if we can add a single value that has more decimal places than what it set in the tool, it should truncate by default. And then, the autoclean Equations could be used. 

For me it does not make sense to manually add `3.0001` to the field if the max decimal places are 2. It should be trimmed by default.

Now, this is different than adding something like `1/3`, which is a static expression that indicates the terms in use.

If the user needs a hack to add `3.0001` then he could enter this expression `30001/10000`

leoheck, 2025-04-04:
Like this? @pierreporte 

```
≈ 3.3 mm (10/3)
≈ 0.1 mm (1/8)
0.1 mm (1/10)
```

pierreporte, 2025-04-04:
@leoheck Exactly. The symbol could be after the value in order to keep numbers aligned in the property view. However, showing the expression is not a good idea because they could be very long. It works for these simple cases though. Why not use the existing expression symbol?

kaiwas, 2025-04-04:
I would also like to mention that zero values ​​after the dot do not necessarily need to be shown https://github.com/FreeCAD/FreeCAD/issues/20490

leoheck, 2025-04-04:
@pierreporte I like to have numbers aligned, indeed. Then, if the symbol is pushed to the end of the symbol, it makes more sense to show an elipsys. (I think).

[image]

leoheck, 2025-04-04:
Hm, Ok, I see. Should we create a different issue for this?
It is deviating a bit from the main topic.

@pierreporte Did you have a chance to read this?

> Now, something different is if we can add a single value that has more decimal places than what it set in the tool, it should truncate by default. And then, the autoclean Equations could be used.

> For me it does not make sense to manually add 3.0001 to the field if the max decimal places are 2. It should be trimmed by default.

> Now, this is different than adding something like 1/3, which is a static expression that indicates the terms in use.

> If the user needs a hack to input 3.0001, then he could enter the expression 30001/10000 or something more visual like 3.0001 * 1

kaiwas, 2025-04-04:
> why not using the ≈ symbol?

this best corresponds to the meaning of rounding.
maybe this symbol as an icon should be displayed on the right side of the field where the expression sign is? so this symbol will not interfere with reading the main value.
well and for it it will be possible to make a tooltip.

pitch-circle, 2025-04-09:
Please could you clarify that by "single numeric value" you mean a unitless quantity.

An Expression with a single unit-ed value is very useful when working in two measurement systems.

For example I can put 1in (one inch) into the Expression. In the property panel I see 25.4mm and also see 1in written in blue.

It would be very bad if this feature were to be lost as a result of this proposal.

In a similar way, any loss of the ability to display an improper fraction - the 10/3 shown above or a more complex fraction like (3+7)/11 - would also be very bad. It is very useful to see the logic behind how a number is constructed in addition to the final answer.

kaiwas, 2025-04-09:
> 1in (one inch)

The imperial system is very specific. I think it always requires a separate logic.

I can't even imagine how you live in this measurement system )))

leoheck, 2025-04-09:
> I can't even imagine how you live in this measurement system )))

Well, this does not matter.

What matters is that if you put anything in the field that changes the unity, it should not be cleaned up, right?

I don't use imperial units, but this does not make them disappear. So, if I am making something for US users, I will use their units because it will be easier for them to understand and find compatible things.

leoheck, 2025-04-09:
@kaiwas the result for the `1 in` the Equation field should be something like this:

```
25.4 mm (1 in)
```

kaiwas, 2025-04-09:
> What matters is that if you put anything in the field that changes the unity, it should not be cleaned up, right?

I understand that. That's why I say that imperial units require their own logic.
This doesn't mean that those who use the metric system of measurements should experience inconvenience for the sake of having fractional inconvenient parts.

FlachyJoe, 2025-04-09:
Related #20724

pitch-circle, 2025-04-11:
> I understand that. That's why I say that imperial units require their own logic.

I am not sure that the unit issue only applies to imperial units.

In many places, FC does unit scaling, so if you write 10000m somewhere, it may be displayed elsewhere as 10km.

Particularly for things that are not simple lengths, there are different conventions used in different fields of use (e.g. stress/pressure - some use kPA, some MPa, some N/mm^2).

Often the Expression field allows the user to write what they understand and what they actually want to see, despite FC's best efforts to convert or re-display that input. It would be a shame to lose this flexibility or to dilute it by tinkering with something that is not fully understood.

kaiwas, 2025-04-11:
> In many places, FC does unit scaling, so if you write 10000m somewhere, it may be displayed elsewhere as 10km.

Oh yeah! I'd like to look the person who came up with this in the eye? One of the worst things you can't turn off