FreeCAD/FreeCAD #20621: gltf export problem in 1.1dev

mnesarco, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

In 1.0 I can export gltf but in 1.1dev it fails:

# 1.0
OS: KDE Flatpak runtime (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * Mnesarco_Utils 0.2.16
  * Marz 0.1.15
  * FreeCAD_Svg 1.0.0.dev5
  * OpenTheme 2024.9.1
  * export_notif
  * fasteners 0.5.37
  * fcapi

# 1.0 Python console

```python
>>> # Gui.Selection.addSelection('Dongle_Case','root')
>>> ### Begin command Std_Export
>>> __objs__ = []
>>> __objs__.append(FreeCAD.getDocument("Dongle_Case").getObject("root"))
>>> import ImportGui
>>> if hasattr(ImportGui, "exportOptions"):
>>>     options = ImportGui.exportOptions(u"/home/mnesarco/Desktop/Dongle-Case-root2.gltf")
>>>     ImportGui.export(__objs__, u"/home/mnesarco/Desktop/Dongle-Case-root2.gltf", options)
>>> else:
>>>     ImportGui.export(__objs__, u"/home/mnesarco/Desktop/Dongle-Case-root2.gltf")
>>> 
>>> del __objs__
>>> ### End command Std_Export
```

It suceeds

# 1.1dev
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41077 (Git)
Build date: 2025/03/30 16:11:00
Build type: Release
Branch: main
Hash: 3a7712bc68a94bab8ea65dfb8f5a2695749c980a
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * FreeCAD_Svg 1.0.0.dev5
  * OpenTheme 2024.9.1
  * export_notif
  * fasteners 0.5.37
  * MarzWorkbench 0.1.15
  * MnesarcoUtils 0.2.15

# 1.1dev Python console

```python
>>> # Gui.Selection.addSelection('Dongle_Case','root')
>>> ### Begin command Std_Export
>>> ## Gui/Application.cpp(833)
>>> __objs__ = []
>>> __objs__.append(FreeCAD.getDocument("Dongle_Case").getObject("root"))
>>> import ImportGui
>>> if hasattr(ImportGui, "exportOptions"):
>>>     options = ImportGui.exportOptions(u"/home/mnesarco/Desktop/Dongle-Case-root2.gltf")
>>>     ImportGui.export(__objs__, u"/home/mnesarco/Desktop/Dongle-Case-root2.gltf", options)
>>> else:
>>>     ImportGui.export(__objs__, u"/home/mnesarco/Desktop/Dongle-Case-root2.gltf")
>>> 
>>> ### End command Std_Export
```

It fails with cryptic error:

```
10:00:30  pyException: Traceback (most recent call last):
  File "<string>", line 6, in <module>
<class 'OSError'>: Cannot save to file: : /home/mnesarco/Desktop/Dongle-Case-root2.gltf
```

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.41077 (Git)
Build date: 2025/03/30 16:11:00
Build type: Release
Branch: main
Hash: 3a7712bc68a94bab8ea65dfb8f5a2695749c980a
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * FreeCAD_Svg 1.0.0.dev5
  * OpenTheme 2024.9.1
  * export_notif
  * fasteners 0.5.37
  * MarzWorkbench 0.1.15
  * MnesarcoUtils 0.2.15
```

### Subproject(s) affected?

File formats

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-04-03:
Guessing it's a fairly recent regression, works OK using:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40759 (Git)
Build date: 2025/03/17 05:00:02
Build type: Release
Branch: (HEAD detached at fd4fd6ccf7)
Hash: fd4fd6ccf7e0316742ebd4235ab6251059d55275
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, IfcOpenShell 0.8.1.post1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/1
Installed mods: 
  * FreeCAD-Ribbon 1.8.1.2 (Disabled)
  * btl 0.9.9
  * Reinforcement
  * FreeCAD-themes 2025.1.7
  * FOSS3DLiveLink.zip
  * DynamicData 2.73.0
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.30.1
  * ChangeAppFont
  * Quetzal 1.3.3
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * Curves 0.6.60
  * FOSS3DLiveLink
  * A2plus 0.4.68
  * ThreadProfile 1.93.0
```

luzpaz, 2025-04-03:
Set this at a 'High' priority since mnesarco's workin on bridging FreeCAD and Blender.

mnesarco, 2025-04-03:
May be this issue related to `-DFREECAD_USE_EXTERNAL_SMESH=OFF` ?
I don't know if gltf exporter requires external smesh

mnesarco, 2025-04-03:
> May be this issue related to `-DFREECAD_USE_EXTERNAL_SMESH=OFF` ? I don't know if gltf exporter requires external smesh

Apparently there is an internal smesh build, so i don't think this is the culprit.

mnesarco, 2025-04-03:
This is my build config btw:

```bash
mkdir -p $BUILD_BASE/build && \
  cd $BUILD_BASE/build && \
  cmake -GNinja -DCMAKE_BUILD_TYPE=Debug \
  -DFREECAD_QT_VERSION=6 \
  -DCMAKE_INSTALL_PREFIX=/usr/local \
  -DFREECAD_USE_PYBIND11=ON \
  -DBUILD_CLOUD=OFF \
  -DBUILD_DRAWING=OFF \
  -DBUILD_JTREADER=OFF \
  -DBUILD_OPENSCAD=OFF \
  -DBUILD_REVERSEENGINEERING=OFF \
  -DBUILD_ROBOT=OFF \
  -DBUILD_SANDBOX=OFF \
  -DBUILD_TEST=OFF \
  -DBUILD_VR=OFF \
  -DENABLE_DEVELOPER_TESTS=OFF \
  -DFREECAD_USE_EXTERNAL_KDL=OFF \
  -DFREECAD_USE_EXTERNAL_SMESH=OFF \
  $BUILD_BASE/FreeCAD
```

Syres916, 2025-04-03:
To quote Werner:

_glTF support is not implemented directly in FreeCAD but is provided by OpenCascade (OCC).
However, if OCC is not built with RapidJSON then its glTF export doesn't work. Unfortunately, from within FreeCAD code it's not possible to determine at build time if OCC's glTF export is active or not. We can only see at runtime that the export failed._

mnesarco, 2025-04-03:
> To quote Werner:
>
> _glTF support is not implemented directly in FreeCAD but is provided by OpenCascade (OCC). However, if OCC is not built with RapidJSON then its glTF export doesn't work. [...]

indeed this is my case:

https://github.com/NixOS/nixpkgs/issues/156681

mnesarco, 2025-04-03:
Ok, I changed my NIX Flake to use a custom build of OCCT with rapidjson. The export gltf kinda works but:

# Exported with 1.0

[image]

# Exported with 1.1dev

[image]

I admit I have no idea what is happening.

mnesarco, 2025-04-03:
Ok, here is: https://github.com/FreeCAD/FreeCAD/issues/18152

belinski-roman, 2025-04-04:
There is a similar problem with STEP export. Only the wireframe representation is exported. It is easy to verify this by importing the STEP back into Freecad. In version 1.0, everything worked as expected

mnesarco, 2025-04-04:
I am looking at the export code, I have found duplicated code to export gltf, step, etc.. and I have no idea which code is actually in use. 

https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Import/App/AppImportPy.cpp
https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Import/Gui/AppImportGuiPy.cpp

mnesarco, 2025-04-04:
Maybe related: https://github.com/FreeCAD/FreeCAD/issues/10861

mnesarco, 2025-04-04:
Using `Import` instead of `ImportGui` module works better, no colors yet but at least no objects missing and not 
transparent.

```
>>> import Import
>>> Import.export(__objs__, u"/home/mnesarco/Desktop/C.gltf")
```

[image]

luzpaz, 2025-04-08:
@mnesarco thanks again for digging into this 🙏

mnesarco, 2025-04-09:
@luzpaz Unfortunately, I can't look into this further right now. Maybe someone with more context about the color change issue—why it was changed from App to Base, how this affects transparency, and why there's duplicated code in Import and ImportGui—can help. I'm not sure which one is supposed to be used, which one is deprecated, or if both have specific use cases. They aren't exactly the same, as I've already shown, but neither sets colors. From my tests, this is the least bad option:

```
>>> import Import
>>> Import.export(objects, "output.gltf")
```

luzpaz, 2025-04-13:
Can anyone weigh-in here ? 👆

closed 2025-04-21 by chennes