FreeCAD/FreeCAD #20622: Flickering with status bar enabled, when passing over the edge of objects

CheezusJeezus, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When one moves their mouse clickly over the edge of a part (passing inside and outside of it) with the status bar enabled, it will cause the toolbars to flicker black. This appears on multiple parts, images, and both the mainline (1.0) and dev branch listed below. 

Bug is entirely cosmetic and does not affect any tools as of yet in my testing, and I don't expect it to. 

Disabling the status bar fixes this bug, so I assume it has to do with the hover description update rate, but I am not a developer. 

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/93.2987
Installed mods: 
  * CfdOF 1.30.1
  * OpenTheme 2024.9.1
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

CheezusJeezus, 2025-04-03:
Applying the opentheme's "reccomended settings" and restarting, seems to have fixed this, so I have closed this, it may have been some kind of setting issue

luzpaz, 2025-04-03:
So this is a stylesheet issue, in other words ?