FreeCAD/FreeCAD #20623: Does App::Feature exist?

aiksiongkoh, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Does App::Feature exist? 
It is used in 
Mod\BIM\nativeifc\ifc_status.py
src\Base\Type.h

There doesn't to be any definition of it.

### Full version info

```shell
In FreeCAD source code.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-04-03:
No it doesn't and it didn't since the first git commit.
It should be `App::GeoFeature`.

Note it's only used in a comment in `src\Base\Type.h`. But the condition is never true in `Mod\BIM\nativeifc\ifc_status.py`
```python
def filter_out(objs):
    """Filter out objects that should not be converted to IFC"""

    nobjs = []
    for obj in objs:
        if obj.isDerivedFrom("Part::Feature"):
            nobjs.append(obj)
        elif obj.isDerivedFrom("Mesh::Feature"):
            nobjs.append(obj)
        elif obj.isDerivedFrom("App::DocumentObjectGroup"):
            if filter_out(obj.Group):
                # only append groups that contain exportable objects
                nobjs.append(obj)
            else:
                print("DEBUG: Filtering out",obj.Label)
        # THIS CODE NEVER RUN
        # elif obj.isDerivedFrom("Mesh::Feature"):
        #    nobjs.append(obj)
        # elif obj.isDerivedFrom("App::Feature"):
        #    if Draft.get_type(obj) in ("Dimension","LinearDimension","Layer","Text","DraftText"):
        #        nobjs.append(obj)
        #    else:
        #        print("DEBUG: Filtering out",obj.Label)
        else:
            print("DEBUG: Filtering out",obj.Label)
    return nobjs
```

FlachyJoe, 2025-04-04:
@maxwxyz why do you close this? I should be solved

maxwxyz, 2025-04-04:
Sorry it seemed like a question to me.

FlachyJoe, 2025-04-04:
@aiksiongkoh can you please re-title?

FlachyJoe, 2025-04-04:
`App::Feature` is also used here: https://github.com/FreeCAD/FreeCAD/blob/9ada83aebd535b9f59763c60fecd24057315857e/src/Tools/bindings/templates/templateModuleAppFeature.py#L32-L37
Imported by `src/Tools/bindings/templates/templateModuleApp.py` but never called because `AppPath + "Features/"` doesn't exist anymore
https://github.com/FreeCAD/FreeCAD/blob/9ada83aebd535b9f59763c60fecd24057315857e/src/Tools/bindings/templates/templateModuleApp.py#L21-L25

This has to be cleaned.