FreeCAD/FreeCAD #20625: Opened files are duplicated multiple times.

leoheck, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

What just happened here?

Is this related to having multiple tabs of the same file opened?

I realized after closing the tool I had Tabs with this `: 1` next to the filename (in the tab name) higher than one. All or some of those tabs were invisible/hidden, and when I closed one trying to check if reopening the file would fix the issue, the other of the same file with a higher index would appear.

[image]

Then, everything started to fail.
For instance, I was trying to double-click the `VarSet` and FreeCad was complaining about something, and it could not show me the content. That was crazy.

The dump happened after I closed FreeCad.
Before, it was just complaining about this:

```
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
2.8e-08 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 
```

I was not doing anything nasty; I was only working on PD features and the only thing I've noticed was that my internet went off. 

This is a terrible bug.
What should I do to help push this fix? 
We cannot work with more than one file when we have such a bug.
This means projects cannot grow in complexity easily.

```

~/Applications …
➜ ./freecad_weekly_40971 
FreeCAD 1.1.0, Libs: 1.1.0devR40971 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

OSHAutoDoc workbench (v0.2.3) loaded
Initializing Assembly4 workbench (0.50.19-dev) ............................. done.
The Sketch has partially redundant constraints!
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Cannot set the datum because the sketch contains conflicting constraints
App.getDocument('Cap').getObject('Sketch409').setDatum(10,App.Units.Quantity('10.000000 mm'))
<Sketch> SketchObject.cpp(9533): External geometry Cap#Sketch373.e11 missing reference: SubtractiveLoft.;#f1f8:11d;:U18;RFI;:H-9f5:9,E
<ElementMap> <string>(1): No hasherRef
No measure handler available for geometry of module: App
No measure handler available for geometry of module: App
<TopoShape> TopoShapeExpansion.cpp(968): hasher mismatch
<TopoShape> TopoShapeExpansion.cpp(968): hasher mismatch
<Sketch> SketchObject.cpp(9533): External geometry Cap#Sketch373.e11 missing reference: SubtractiveLoft.;#f1f8:11d;:U18;RFI;:H-9f5:9,E
ControlSingleton::showDialog: Can't show  Gui::TaskView::TaskDialog  since there is already an active task dialog
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
2.8e-08 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
30.3563 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
30.7275 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
36.7111 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
45.0675 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
51.395 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
52.6202 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 
<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
[... 117 lines cut ...]
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-11-21:
Yes, this still happens.

FEA-eng, 2026-02-20:
@leoheck I assume it's still relevant, but a minimum file/steps would be crucial.

FEA-eng, 2026-09-12:
See [this](https://github.com/FreeCAD/FreeCAD/issues/20625#issuecomment-3932689076)

luzpaz, 2026-10-04:
Hard to grok how to trigger this bug. @leoheck do you remember how to do so (and can you provide detailed steps) ?