FreeCAD/FreeCAD #20626: Export to DXF using Techdraw issue

jmjack73, 2025-04-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using TechDraw, if the part view is changed using the down arrow or rotate angles, it will not output a clean dxf file that can be opened in Adobe Illustrator. Illustrator displays error 2067 when attempting to open the file. If the part view is left alone and not moved using the direction arrows, the export is clean, and Illustrator successfully opens the DXG file.  The export issue is also consistent with the exported file size being 1614KB.  It also appears using the up arrow of the side arrows will fix something and the dxf ouput is fine. Please look into this. 

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

WandererFan, 2025-04-04:
Any chance of a sample file and/or picture?  Having trouble visualizing this.

kaiwas, 2025-04-05:
I can mention that notes (text) and balloons are not exported to dxf. Perhaps there are also problems with the sizes.

luzpaz, 2025-05-07:
> Any chance of a sample file and/or picture? Having trouble visualizing this.

@jmjack73

closed 2025-12-04 by github-actions[bot]

kaiwas, 2025-12-04:
> didn't find a ticket for these. Should we open ?

I made a small test DXF file. (Remove the .zip extension.)

[Untitled-1.dxf.zip](https://github.com/user-attachments/files/23941196/Untitled-1.dxf.zip)

It seems my comments are no longer relevant.
The screenshots show exporting shapes and text from Corel Draw to .dxf.

[image]

Importing to version 0.21

[image]

[image]

Importing to 1.1RC1

[image]

I don't have any other files (I don't usually work with DxF). But at first glance, the import has improved significantly.

(Import in autoCAD)

[image]