FreeCAD/FreeCAD #20632: Toobar is corrupted

ReneSci, 2025-04-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

```
FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311.7z
```
In the latest Windows build, the toolbar is not working:

[image]

I can not unfold the toobars...

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US) [ OS: Danish/Denmark (da_DK) ]
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Classic/
Logical DPI/Physical DPI/Pixel Ratio: 96/141.951/1
Installed mods: 
  * A2plus 0.4.65
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * MeshRemodel 1.9.23
  * OpenDark 2024.3.13
  * OpticsWorkbench 1.0.18
  * QuickMeasure 2022.10.28
  * sheetmetal 0.7.3
  * workfeature
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ReneSci, 2025-04-04:
It seems like it is the theme "OpenDark" which makes this error.

[image]

Changing to "FreeCAD Dark" works

[image]

maxwxyz, 2025-04-04:
please open an issue in the OpenTheme repo then,

closed 2025-04-04 by maxwxyz