FreeCAD/FreeCAD #20633: Part: findPlane regression

Roy-043, 2025-04-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The findPlane method fails to find a plane for faces that are obviously flat and changing the tolerance does not have any effect. This is a regression introduced in V1.0 AFAICT.

1. Open the attached file.
2. Select one of the slanted faces.
3. Invoke [Std_SendToPythonConsole](https://wiki.freecad.org/Std_SendToPythonConsole).
4. Paste this in the Python console: `sub.findPlane()`
5. Result: no plane is returned.
6. Try again with a very high tolerance: `sub.findPlane(1)`
7. Result: again no plane is returned.

Note: Using a Linux version of the latest dev as the Windows version will not start on Windows 8.

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```

### Subproject(s) affected?

Part

### Anything else?

Problem does not occur in V021:
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-04-05:
The regression was introduced between mid May 2024 and early July 2024:

Good:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git) AppImage
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
```

Bad:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
```

Roy-043, 2025-04-05:
Pinpointing it a bit more (both from 2nd half of May 2024)

Good:
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37358 (Git)
Build type: Release
Branch: main
Hash: 3a0152b076dd7791c9724c286963d74b5814e24f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

Bad:
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37436 (Git)
Build type: Release
Branch: main
Hash: 040502280fc9637fb3aada037523283168b81a02
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:

```

Syres916, 2025-04-05:
@Roy-043 that feedback isn't good as it looks like just when the _TNP_ was fully enabled but more concerning is that the version of _OCC_ looks to be a factor as well. I'll have another go with Bisecting a vm that usesOCC  7.7.2 tomorrow, I've been using an old system with OCC 7.3.3 this evening and it doesn't correlate with your Bad build.

Syres916, 2025-04-06:
I can confirm using OCC 7.6.3 and older I can use the `sub.findPlane()` successfully even with current main so it looks like there's been an oversight with the enabling of TNP and use of OCC 7.7.x and above, don't know how long it'll take to track down.

maxwxyz, 2025-06-06:
https://github.com/wwmayer/FreeCAD/commit/03c589fb124cbdbd64df796d5f0f911439427a6d

luzpaz, 2025-06-18:
What's the verdict here ?

luzpaz, 2025-07-12:
So is this a toponaming bug or a regression caused by the TNP code ?

Roy-043, 2025-07-13:
Judging from the info in Werner's PR it is an OCC version issue.

pieterhijma, 2025-08-31:
I can confirm the problem and that Werner's fix works. Without Werner's fix, both `sub.findPlane()` and `sub.findPlane(1)` don't return a plane. With Werner's fix, `sub.findPlane()` doesn't return a plane while `sub.findPlane(1)` does return a plane. I recommend to incorporate Werner's commit.

stefano2734, 2026-09-19:
I do not see here the clear solved occt version of this problem. 
Or there wasn’t a problem with TNP in newer Versions at occt 7.7.x.