FreeCAD/FreeCAD #20635: Helix Negative Cone Angle changes the grow orientation.

leoheck, 2025-04-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am trying to make a taper angle with a PD Helix using the `Cone Angle`.
However, having negative values changes the growth orientation. 

https://github.com/user-attachments/assets/560db3c3-46df-4079-a97c-a3fe592d72a5

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.7
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

possebaer, 2025-04-06:
+1
Experienced the same today, and searched here if issue is known/resolved. I experienced this issue with the 1.0 version on openSUSE.

luzpaz, 2025-04-06:
@possebaer full [about]([url](http://wiki.freecad.org/About)) info please, TIA

possebaer, 2025-04-06:
Here we go:

OS: openSUSE Tumbleweed (KDE/plasma6/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 5.15.16, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
  * Dracula 0.0.5
  * boltsfc 2022.11.5
  * Curves 0.6.53
  * Render 2024.12.15
  * A2plus 0.4.68
  * Freecad-Built-in-themes-beta 1.2.2
  * ThreadProfile 1.92.0
  * CurvedShapes 1.0.13
  * Behave-Dark-Colors 0.1.1
  * toSketch 1.0.1
  * OpenDark.backup1733237432.6359558 2024.3.13 (Disabled)

possebaer, 2025-04-06:
Seems to be fixed in latest weekly build on Linux (AppImage):

OS: openSUSE Tumbleweed (KDE/plasma6/xcb)
Architecture: x86_64
Version: 1.1.0dev.41138 (Git) Conda AppImage
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/34.735/1
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
  * Dracula 0.0.5
  * boltsfc 2022.11.5
  * Curves 0.6.53
  * Render 2024.12.15
  * A2plus 0.4.68
  * Freecad-Built-in-themes-beta 1.2.2
  * ThreadProfile 1.92.0
  * CurvedShapes 1.0.13
  * Behave-Dark-Colors 0.1.1
  * toSketch 1.0.1
  * OpenDark.backup1733237432.6359558 2024.3.13 (Disabled)

I opened my file there, and the helix subtraction with cone angle works as expected (negative sign does not change direction). So it seems it was fixed between git 40971 and git 41138.

possebaer, 2025-04-06:
@leoheck: Could you try that weekly build as well and see if it fixes for you too?

FlachyJoe, 2025-04-06:
[#18108#issuecomment-2576345773](https://github.com/FreeCAD/FreeCAD/issues/18108#issuecomment-2576345773) ?

possebaer, 2025-04-06:
@leoheck  Ok, interesting. 

My original issue was with a helix that was subtracted from a part, and there the  change in cone angle sign had the same effect that is described by your bug report. Then I updated to current weekly (note: mine is 41138, yours is a few commits behind -- not sure if that is noteworthy), and this problem is gone for me. 

So with the weekly I  tried to reproduce your last example, and I can reproduce that -- that's not fixed.

I then created a (stupid) part and added your helix, and then the effect is gone -- I can have a positive and a negative cone angle here: 

[image]

[image]

I does not really matter if I use reverse direction or not, but inside the part the helix is not very well visible....

So the problem seems to be fixed if the helix is attached to some other object, but not if it's on its own.

possebaer, 2025-04-06:
@leoheck @FlachyJoe  Dang, I didn't read the linked bug report carefully enough, yes it's only a problem in this case if the center of the triangle is on an axis. I could recreate that. @leoheck example works if the triangle is moved away from the axis.

leoheck, 2025-04-06:
Fuck, these weekly versions are crazy. They are the same commit version for weeks, and now today it has changed 2 or 3 times, and the date of each package in the freecad-bundle is still showing it was changed 2 days ago. I just downloaded this version 41127 this morning. This this thing is horrible.

leoheck, 2025-04-06:
> I does not really matter if I use reverse direction or not, but inside the part the helix is not very well visible....

@possebaer toggle the transparency to check the helix inside of the part

FlachyJoe, 2025-04-12:
Can someone confirm it's not a duplicate of #18108?

possebaer, 2025-04-12:
No, my experience would suggest this is a duplicate. I quickly checked with the file of @leoheck, and with my recent weekly build of FreeCAD, it is sufficient to move the triangle center point away from the Axis to fix the behaviour, as described in the referenced bug report. 
If no further info pops up, I think this should be marked a duplicate. 

[](https://github.com/FreeCAD/FreeCAD/issues/18108#issuecomment-2576345773)

My version information:

OS: openSUSE Tumbleweed (KDE/plasma6/xcb)
Architecture: x86_64
Version: 1.1.0dev.41138 (Git) Conda AppImage
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: https://github.com/FreeCAD/FreeCAD/commit/cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/34.735/1
Installed mods:

    ProDarkThemePreferencePack 1.0.0
    Dracula 0.0.5
    boltsfc 2022.11.5
    Curves 0.6.53
    Render 2024.12.15
    A2plus 0.4.68
    Freecad-Built-in-themes-beta 1.2.2
    ThreadProfile 1.92.0
    CurvedShapes 1.0.13
    Behave-Dark-Colors 0.1.1
    toSketch 1.0.1
    OpenDark.backup1733237432.6359558 2024.3.13 (Disabled)

luzpaz, 2025-04-17:
@leoheck do you also agree this is a dupe of @18108  ?

leoheck, 2025-04-18:
It looks like the same issue. But the other one is a bit hard to understand.

I am stating that the direction of the growth should not change due to the cone angle.
For that, we have the flag `Reversed`.

The cone angle should describe if the end of the thread is getting far from the center (cone angle > 0) or close to the center (cone angle < 0)

FlachyJoe, 2025-04-18:
@leoheck I agree it's a bug but I can't find it in the FreeCAD sources: the path to sweep the profile is well generated at the good place but the result shape is wrong if the profile is too close the axe. 
It is an OCC bug or it is an error with a precision set. I can't go further.

Feel free to add comments to clarify #18108 and close this issue.

closed 2025-04-18 by luzpaz