FreeCAD/FreeCAD #20637: UI/UX: Add multiples properties to a VarSets

leoheck, 2025-04-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The checkbox for adding a new property is kind of annoying.
It also needs to be unchecked on the last item.

[image]

Please replace this with a new button with this option.
If the user wants a new value, he will use that button.

Also, hotkeys (and tooltips for the buttons/hotkeys)
Enter would apply the Settings and Leave.
Something like `Ctrl+Enter` or `Shift+Enter` or something else.

Example... 

[image]

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.2
  * lattice2 1.0.0
  * kicadStepUpMod 11.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-04-05:
It is unlikely that this proposal is consistent with the overall concept.
Adding new "objects" or properties is required in many places. Most often, this looks like a "continue" checkbox.

If your proposal is implemented, then for the sake of uniformity it should be done everywhere.

leoheck, 2025-04-05:
Ah, this explains why this VarSet popup is so bad to use. It has to follow the standard of the other places.

A simple button would so much better for usability.

kaiwas, 2025-04-05:
> Ah, this explains why this VarSet popup is so bad to use. It has to follow the standard of the other places.
> 
> A simple button would so much better for usability.

I generally agree with you. But it would be more correct to strive for uniformity.

Now we often get confused anyway. Somewhere there is OK and a checkbox "continue", somewhere "apply". Somewhere just "Close".
You can go crazy.

I would be glad to see simplification in the work of any tools, but it would be desirable that the concept worked similarly everywhere. If you fix it, then fix it everywhere. Otherwise, the inconsistencies accumulate.

kaiwas, 2025-04-05:
> I agree. Can we fix it every where then? Because, similarly, the input filds doesn't behave the same everywhere too.

I am not a developer )

Perhaps developers might be interested in this if you make a thoughtful proposal with a description of all the points (What, where and how should change). But it seems to me that this is a titanic work on analyzing the existing inconsistencies in usability. It seems that the design group is doing something similar, but the possibilities are still limited.

maxwxyz, 2025-04-06:
@pieterhijma FYI

leoheck, 2025-04-07:
It could be something like `Next` and the Tooltip `Apply and create a new variable.`

furgo16, 2025-04-28:
If this helps as a data point, this is how it's done in GitHub when adding multiple issue reports:

[image]

maxwxyz, 2025-04-28:
@pieterhijma I'm in favor of the checkbox instead of another button. The checkbox should remember its last state so clicking OK will keep creating new ones until unchecked. Similar as the Draft commands to draw new geometry.

leoheck, 2025-04-28:
@pieterhijma the next button will keep the state forever. The checkbox is a really poor choice. The user just has use the next button to create a new variable, otherwise it will to use the apply, or ok, or done button.

pierreporte, 2025-04-28:
In Solidworks, all parameters are listed in a dialog box. There is one empty line, to be used to add a new parameter. It reappears at the bottom each time a new parameter is created. I quite like this UI. It could be used in place of the property view: because we consider it an advanced feature, this new dialog could be used instead. Maybe it could be a task panel.

[image]

leoheck, 2025-04-28:
Bah, honestly, this is horrible. The way it is now is simple and quick. 

There are some usage tweaks that still are annoying like those interruptions when the user doesn't fill the fields the Way the developer wanted.

Why the hell they made that? But in general, this popup as it is, is simple, quick and effective. The checkbox should gone to have a quicker usage, including shortcuts to click the done and next buttons. And that's it.

Also, there are other things that have to be improved. For instance when we want to create a new variable when applying a constrain in the Sketcher. The window increases but the size of it is messed up. Specially when disabling the checkbox to show VarSet creation.

pieterhijma, 2025-04-30:
I found the comment that suggested an "Add next" button: https://github.com/FreeCAD/FreeCAD/issues/14633#issuecomment-2171585064.

I have no strong preference, I will simply implement what the community wants. Since there are diverging preferences, I'm going to ask the @FreeCAD/design-working-group. DWG, could you please advise on whether we want a checkbox for adding a next property or a separate button as is suggested here?

And in your reply, could you please let us know that the decision you made that this is the consensus of the DWG? Thank you very much in advance.

kaiwas, 2025-04-30:
> I found the comment that suggested an "Add next" button: [#14633 (comment)](https://github.com/FreeCAD/FreeCAD/issues/14633#issuecomment-2171585064).

Personal opinion.
There are a lot of "different" things in the program now. The same similar actions often have different buttons.
Somewhere it's "OK/Apply/Cancel", somewhere it's "OK/Close". Somewhere it's just "Close" ))

All this is very bad from the point of view of system design.
I have nothing against the "Add Next" button. But it will add another entity that will not be consistent with the other actions.

Maybe it could be a combination of "OK/Apply/Cancel"?
OK - adds a value and closes the panel.
Apply - adds a value and resets the input fields for adding the next one (analogous to "add next")
Cancel - closes the window without an action.

MisterMakerNL, 2025-04-30:
> If this helps as a data point, this is how it's done in GitHub when adding multiple issue reports:
> 
> [image]

I am with this, but the checkbox has to be next to the button exactly like example.
If you just add a next button, you also have to tell the user the current is saved, otherwise the user might expect a back button too to inspect the vars they inputted. If you do it like Github does it is clear that you made the car and you are just going to make another one.

pieterhijma, 2025-04-30:
So, @kaiwas, essentially, we can ask the @FreeCAD/design-working-group as well if their advice can take overall consistency in FreeCAD in account as well. That seems like a good idea.

kadet1090, 2025-04-30:
> > Maybe it could be a combination of "OK/Apply/Cancel"?
> > OK - adds a value and closes the panel.
> > Apply - adds a value and resets the input fields for adding the next one (analogous to "add next")
> > Cancel - closes the window without an action.
> 
> Yes, this is an improved UI/UX compared to a checkbox.

Ok, Apply and Cancel have very precise meaning:

> "Ok" is "Apply changes and close dialog"
> "Apply" is "Apply changes and keep the dialog open"
> "Cancel" is "Cancel non applied changes"

For "Apply" in that case the typical assumption would be that it saves and adds the property, indeed, but does not reset state of the dialog to add new property. It is abusing the notion of "Apply". And yes, it is done really often within FreeCAD but we should not make more of the same problem. As meaning of "Apply" in that case is not clear, it really should just be "Add and create next" (or something like that) - that is unambiguous and should be pretty clear for everyone.

In discussion we also noted that "Cancel" meaning with "Apply" present is a bit ambigous and should be avoided as it is unclear if it should reset changes to state before dialog was open or to last applied position. In second case simple "Close" should be preferred. "Ok" and "Cancel" is an ok pair.

Mannshoch, 2025-05-01:
I prefer a kind of Property table. where you could see what already exist and may even help to reflect better what you have and what you need. In the solidworks equation manager you even be able, if you currently entering an equation, to click on other equation to add their property name into your current equation. I used that often.

leoheck, 2025-05-01:
We should stop talking about copying solidworks, right? There are many other cad tools but Freecad doesn't have to be them.

MisterMakerNL, 2025-05-01:
> We should stop talking about copying solidworks, right? There are many other cad tools but Freecad doesn't have to be them.

To be honest I used like 90% of all the big CAD software programs professional, and SolidWorks by far is the best in UI.

kadet1090, 2025-05-01:
> We should stop talking about copying solidworks, right? There are many other cad tools but Freecad doesn't have to be them.

We are not talking about copying any particular software - we are not making a clone. However, doing research over what software in the same category does is very important step in UX processes. Stand on the shoulder of giants you may even say. Other CAD suites already did solve many problems, including that one, in some way. Better or worse, but they did - we should learn from them - avoid their mistakes and get inspired by what we think is good.

Furthermore, this is even codified as so called Jakob's law (or 4th Nielsen's Heuristic):

> Users should not have to wonder whether different words, situations, or actions mean the same thing. Follow platform and industry conventions.

If all CADs work in the same (or similar) way - odds are that users are used to it and know it by hearth so it is worth following the convention. If FreeCAD wants to have UX that is perceived as good by the users it must follow standard practices from other cads when applicable. Trying to reinvent wheel is one of biggest UX mistakes often and we should avoid it.

leoheck, 2025-05-01:
The research is indeed important. But I disagree with the conclusion, otherwise it should be copied when it was being designed. The way this feature is now is good, it is just have to be tweaked a bit to be awesome. There are so many other important features to implement. For instance, now we cannot rename properties, and this is the real issue. Then instead of working on that, we are discussing a window that is already good enough at least in a short term. The real issue that has to be fixed (renaming properties) is being addressed in a macro but this is something that must be the core of this VarSet feature.

leoheck, 2025-05-01:
> I used like 90% of all the big CAD software

Maybe this is the problem, you are biased by the other tools. And this explains why the whole Freecad have many different ways of doing the same thing spread along different workbenches as @kaiwas always say.

> So, [@kaiwas](https://github.com/kaiwas), essentially, we can ask the @FreeCAD/design-working-group as well if their advice can take overall consistency in FreeCAD in account as well. That seems like a good idea.

This is the most important step. Having guidelines on how to implement UI/UX, so devs will have a standard and right way to implement things instead of just copying any other random software when they want.

PR #23426 (2025-09-13): Gui: Reconcile Add Property Dialogs
This PR removes the old "Add Property" dialog in favor of the "Add Property" dialog of VarSets.

The VarSet dialog is now generic and works for any document object or document. In documents, you don't have to select a property any longer to add a new property.

Please note that other than the old "Add Property" dialog, this dialog does not support adding properties to multiple document objects. Since this was an obscure feature (see https://github.com/FreeCAD/FreeCAD/issues/23424) and allowing adding values directly makes the logic more complicated, I believ [... cut]
closed 2025-09-13 by kadet1090