FreeCAD/FreeCAD #20641: Crash on export a boolean operation

codertao, 2025-04-04:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Attempt to export a .step file of a body that is a boolean operation results in crash, with stack trace:

Witnessed twice on different freecad projects, only tested with boolean cut; doing an operation on the boolean (chamfering an edge) allows export to complete successfully. 

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7fb3c58be520]
#1  /tmp/.mount_FreeCAGiMGbd/usr/lib/././libTKXDESTEP.so.7.7(+0x89291) [0x7fb37e1e7291]
#2  /tmp/.mount_FreeCAGiMGbd/usr/lib/././libTKXDESTEP.so.7.7(+0x8a078) [0x7fb37e1e8078]
#3  0x7fb37e1e99e1 in STEPCAFControl_Writer::writeColors(opencascade::handle<XSControl_WorkSession> const&, NCollection_Sequence<TDF_Label> const&) from /tmp/.mount_FreeCAGiMGbd/usr/lib/././libTKXDESTEP.so.7.7+0x9e1
#4  0x7fb37e206dff in STEPCAFControl_Writer::transfer(STEPControl_Writer&, NCollection_Sequence<TDF_Label> const&, STEPControl_StepModelType, char const*, bool, Message_ProgressRange const&) from /tmp/.mount_FreeCAGiMGbd/usr/lib/././libTKXDESTEP.so.7.7+0x17ef
#5  0x7fb37e20726f in STEPCAFControl_Writer::Transfer(opencascade::handle<TDocStd_Document> const&, STEPControl_StepModelType, char const*, Message_ProgressRange const&) from /tmp/.mount_FreeCAGiMGbd/usr/lib/././libTKXDESTEP.so.7.7+0x20f
#6  0x7fb37e7b5117 in Import::WriterStep::write(opencascade::handle<TDocStd_Document>) const from /tmp/.mount_FreeCAGiMGbd/usr/lib/./Import.so+0x137
#7  /tmp/.mount_FreeCAGiMGbd/usr/lib/ImportGui.so(+0x33eaa) [0x7fb37ef47eaa]
#8  /tmp/.mount_FreeCAGiMGbd/usr/lib/ImportGui.so(+0x2f32d) [0x7fb37ef4332d]
#9  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libFreeCADBase.so(method_keyword_call_handler+0x374) [0x7fb3c709a774]
#10  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libpython3.11.so.1.0(+0x20603d) [0x7fb3c7ab803d]
#11  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7fb3c7a97b5c]
#12  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7fb3c7aa3532]
#13  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7fb3c7b5dd1d]
#14  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7fb3c7b5d794]
#15  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7fb3c7b7cbf3]
#16  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7fb3c7b7869a]
#17  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7fb3c7b6c9c9]
#18  0x7fb3c710441c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libFreeCADBase.so+0x6c
#19  0x7fb3c8431ba4 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libFreeCADGui.so+0x94
#20  0x7fb3c8323646 in Gui::Application::exportTo(char const*, char const*, char const*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libFreeCADGui.so+0x526
#21  0x7fb3c844e514 in StdCmdExport::activated(int) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libFreeCADGui.so+0x694
#22  0x7fb3c843b59f in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libFreeCADGui.so+0x23f
#23  0x7fb3c843b98f in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libFreeCADGui.so+0x12f
#24  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7fb3c5faf6f2]
#25  0x7fb3c6a9f462 in QAction::triggered(bool) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Widgets.so.5+0x32
#26  0x7fb3c6aa1e2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Widgets.so.5+0xab
#27  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Widgets.so.5(+0x2e12e2) [0x7fb3c6c162e2]
#28  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Widgets.so.5(+0x2e8c34) [0x7fb3c6c1dc34]
#29  0x7fb3c6ae3c08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Widgets.so.5+0x178
#30  0x7fb3c6aa53ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#31  0x7fb3c6aad262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#32  0x7fb3c83e4559 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libFreeCADGui.so+0xc9
#33  0x7fb3c5f7ab28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Core.so.5+0x108
#34  0x7fb3c6aab3eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#35  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Widgets.so.5(+0x1c72a5) [0x7fb3c6afc2a5]
#36  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7fb3c6aff0c9]
#37  0x7fb3c6aa53ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#38  0x7fb3c83e4559 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libFreeCADGui.so+0xc9
#39  0x7fb3c5f7ab28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Core.so.5+0x108
#40  0x7fb3c64073ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Gui.so.5+0x84c
#41  0x7fb3c63dd0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Gui.so.5+0xac
#42  /tmp/.mount_FreeCAGiMGbd/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7fb3bfbd3bea]
#43  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7fb3c1f3fad7]
#44  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7fb3c1f42ac7]
#45  /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7fb3c1f430ce]
#46  0x7fb3c5fcea36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Core.so.5+0x66
#47  0x7fb3c5f7970b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Core.so.5+0x11b
#48  0x7fb3c5f81212 in QCoreApplication::exec() from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libQt5Core.so.5+0x82
#49  0x7fb3c8343744 in Gui::Application::runApplication() from /tmp/.mount_FreeCAGiMGbd/usr/bin/../lib/libFreeCADGui.so+0x8a4
#50  /tmp/.mount_FreeCAGiMGbd/usr/bin/freecad(+0x78b7) [0x55a0a59258b7]
#51  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fb3c58a5d90]
#52  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fb3c58a5e40]
#53  /tmp/.mount_FreeCAGiMGbd/usr/bin/freecad(+0x7c2a) [0x55a0a5925c2a]
```

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Anything else?

[shift_knob.zip](https://github.com/user-attachments/files/19608134/shift_knob.zip)
Doing an export to step on the chamfer works; doing an export on the boolean crashes after the popup for export options.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

jonzirk76, 2025-04-05:
This crash/workaround happens on my system as well

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/

```

luzpaz, 2025-04-05:
Please consider testing on dev version[^1]

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

codertao, 2025-04-05:
Also seeing it on current weekly build; stack trace:

```
FreeCAD 1.1.0, Libs: 1.1.0devR41138 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f094cf9b520]
#1  /tmp/.mount_FreeCAfbBmaN/usr/lib/././libTKDESTEP.so.7.8(+0x39ae3e) [0x7f08dddc5e3e]
#2  /tmp/.mount_FreeCAfbBmaN/usr/lib/././libTKDESTEP.so.7.8(+0x39be2d) [0x7f08dddc6e2d]
#3  0x7f08dddc95b6 in STEPCAFControl_Writer::writeColors(opencascade::handle<XSControl_WorkSession> const&, NCollection_Sequence<TDF_Label> const&) from /tmp/.mount_FreeCAfbBmaN/usr/lib/././libTKDESTEP.so.7.8+0xc66
#4  0x7f08ddde3c70 in STEPCAFControl_Writer::transfer(STEPControl_Writer&, NCollection_Sequence<TDF_Label> const&, STEPControl_StepModelType, char const*, bool, Message_ProgressRange const&) from /tmp/.mount_FreeCAfbBmaN/usr/lib/././libTKDESTEP.so.7.8+0x1260
#5  0x7f08ddde40c8 in STEPCAFControl_Writer::Transfer(opencascade::handle<TDocStd_Document> const&, StepData_ConfParameters const&, STEPControl_StepModelType, char const*, Message_ProgressRange const&) from /tmp/.mount_FreeCAfbBmaN/usr/lib/././libTKDESTEP.so.7.8+0x2d8
#6  0x7f08ddde4234 in STEPCAFControl_Writer::Transfer(opencascade::handle<TDocStd_Document> const&, STEPControl_StepModelType, char const*, Message_ProgressRange const&) from /tmp/.mount_FreeCAfbBmaN/usr/lib/././libTKDESTEP.so.7.8+0xb4
#7  0x7f0888177eb8 in Import::WriterStep::write(opencascade::handle<TDocStd_Document>) const from /tmp/.mount_FreeCAfbBmaN/usr/lib/./Import.so+0x128
#8  /tmp/.mount_FreeCAfbBmaN/usr/lib/ImportGui.so(+0x32dd2) [0x7f08890a3dd2]
#9  /tmp/.mount_FreeCAfbBmaN/usr/lib/ImportGui.so(+0x2e45d) [0x7f088909f45d]
#10  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libFreeCADBase.so(method_keyword_call_handler+0x374) [0x7f094ea87be4]
#11  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libpython3.11.so.1.0(+0x1fae5d) [0x7f094f4d4e5d]
#12  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x2b4) [0x7f094f4b3b84]
#13  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6e4) [0x7f094f4bf314]
#14  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libpython3.11.so.1.0(+0x29ee6d) [0x7f094f578e6d]
#15  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f094f578864]
#16  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libpython3.11.so.1.0(+0x2bce33) [0x7f094f596e33]
#17  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libpython3.11.so.1.0(+0x2b8a4a) [0x7f094f592a4a]
#18  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7f094f586f29]
#19  0x7f094eaf9508 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libFreeCADBase.so+0x68
#20  0x7f094fe86164 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libFreeCADGui.so+0x94
#21  0x7f094fd6b7f7 in Gui::Application::exportTo(char const*, char const*, char const*) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libFreeCADGui.so+0x547
#22  0x7f094fea2f27 in StdCmdExport::activated(int) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libFreeCADGui.so+0x747
#23  0x7f094fe8f17f in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libFreeCADGui.so+0x21f
#24  0x7f094fe8f52f in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libFreeCADGui.so+0x12f
#25  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Core.so.6(+0x30a527) [0x7f094d6ab527]
#26  0x7f094df24583 in QAction::triggered(bool) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Gui.so.6+0x43
#27  0x7f094df27972 in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Gui.so.6+0xf2
#28  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Widgets.so.6(+0x3c2275) [0x7f094e5ef275]
#29  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Widgets.so.6(+0x43c452) [0x7f094e669452]
#30  0x7f094e7e0d94 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Widgets.so.6+0xa64
#31  0x7f094e7d10f4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Widgets.so.6+0x84
#32  0x7f094e7d1f8d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Widgets.so.6+0xddd
#33  0x7f094fe317c9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libFreeCADGui.so+0xc9
#34  0x7f094d6ee29a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Core.so.6+0x16a
#35  0x7f094e7d9398 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Widgets.so.6+0x1e8
#36  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Widgets.so.6(+0x5b232e) [0x7f094e7df32e]
#37  /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Widgets.so.6(+0x5b5930) [0x7f094e7e2930]
#38  0x7f094e7d10f4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Widgets.so.6+0x84
#39  0x7f094fe317c9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libFreeCADGui.so+0xc9
[... cut ...]
#49  0x7f094d726c0a in QCoreApplication::exec() from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libQt6Core.so.6+0x9a
#50  0x7f094fd8cf34 in Gui::Application::runApplication() from /tmp/.mount_FreeCAfbBmaN/usr/bin/../lib/libFreeCADGui.so+0x734
#51  /tmp/.mount_FreeCAfbBmaN/usr/bin/freecad(+0x5ca0) [0x55622245cca0]
#52  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f094cf82d90]
#53  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f094cf82e40]
#54  /tmp/.mount_FreeCAfbBmaN/usr/bin/freecad(+0x602a) [0x55622245d02a]
```

codertao, 2025-04-05:
Export to STL works, to STEP or STEPZ does not; I'm guessing it's something specific to step / writeColors.

chennes, 2025-09-25:
@pieterhijma can you test with OCCT 7.9?

NSUBB, 2025-10-05:
I cannot reproduce a crash. 
I tried exporting:

1. The Part with & without the Chamfer feature deleted
2. The Body with & without the Chamfer feature deleted
3. The Chamfer Tip feature selected (bad practice) instead of the Body
4. The Boolean feature selected (bad practice) instead of the Body with & without the Chamfer feature deleted. 

Even though no crashes, I did reproduce the strange T-shape (appears to be the overall shape hollow) that FEA-eng got. That was produced by exporting the PartDesign body Boolean feature directly (bad practice) instead of exporting the body itself.

`OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * MeshRemodel 1.10.38
  * Telemetry 1.0.2
`

chennes, 2025-10-18:
@codertao can you still reproduce this?

FEA-eng, 2026-10-17:
Now I got an empty step file after exporting the boolean.

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261013 (Git shallow)
Build date: 2026/10/13 19:55:30
Build type: Release
Branch: (HEAD detached at 50daee777)
Hash: 50daee777321b5085accd1f57b57f1a7c1781cae
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 617.14/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```

maxwxyz, 2026-11-07:
Part of this is fixed in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/dac4818a5a58aca5d165916274bb2def392cdfd3

It exports the Boolean result instead of its tool bodies, which addresses the empty or wrong export reported later in this thread. I cannot say this fixes the original writeColors crash.