FreeCAD/FreeCAD #20644: New cmake error pinpointed to PR20606

Syres916, 2025-04-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Having got compiling to work again with two additional switches at the beginning of the week today a new error has appeared which I've pinpointed to https://github.com/FreeCAD/FreeCAD/commit/7986ea0cba326787b69d481a79c2750a09c5a230
Therefore in order to compile I've reverted that commit.

[New_cmake_Error.txt](https://github.com/user-attachments/files/19615057/New_cmake_Error.txt)

The cmake command that is being used is:

`cmake -DOCCT_CMAKE_FALLBACK=1 -DCMAKE_BUILD_TYPE=Release -DFREECAD_QT_VERSION=6 -DFREECAD_USE_OCC_VARIANT="Official Version" -DFREECAD_USE_PCL=Off -DENABLE_DEVELOPER_TESTS=Off -DCMAKE_POLICY_VERSION_MINIMUM=3.5 ../freecad-daily-source`

### Full version info

```shell
https://github.com/FreeCAD/FreeCAD/commit/7986ea0cba326787b69d481a79c2750a09c5a230
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-04-05:
FYI @tritao

maxwxyz, 2025-04-06:
So can this issue be closed or does it need a fix in the code?

Syres916, 2025-04-06:
@maxwxyz I'll close this even though I'm in the middle of a bisect from last May but please raise this at the merge meeting that this is now the 3rd PR in as many weeks that has broken compiling with nothing obvious in the PR, if the author knew what was going to happen it should be clearly stated. Just for instance, perfectly reasonable chance that in six months time if someone if bisecting for another regression around the end of March/early April they are going to be hit with this and will it be obvious to them why????

closed 2025-04-06 by Syres916