FreeCAD/FreeCAD #20646: [Draft/Bim] Add the "Geometry Link" checkbox to the "Dimensions" tool

kaiwas, 2025-04-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Yes, at the moment, all dimensions are created without reference to geometry. That is, when we change an object, the dimensions are not rebuilt.

We can manually add a link to geometry, but this is very labor-intensive if there are a lot of dimensions. We will have to select each one and configure the properties of each dimension.

If you add the "auto-link to geometry" checkbox, this process will be greatly accelerated.

I know that this setting is highly dependent on toponymy. You can write some kind of warning next to the checkbox.

An approximate concept is in the picture.

[image]

A demonstration of the current version in the video. You can see how many actions have to be performed for just one dimension.

1) I create a dimension.
2) I search for the required field in the property
3) I re-select the measured object.

https://github.com/user-attachments/assets/60971f2e-193f-4dc3-a489-d0bb04114de4

This checkbox would significantly speed up the process.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41138 (Git) Conda AppImage
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Test
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.60
  * freecad.gears 1.3.0
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-04-05:
If you select edge, instead of vertex dimensions are auto updating themselves but there is problems with this appoarch.For example,
when you make continous dimensions you have to select vertexesand  dimensions are not updating. For auto-updated dimensions you have to select edges one by one, but this doesn't work when you have lots of dimensions like architectural plans because firstly you can't select window sizes(window gap in wall like image) correctly  and you have to align dimesnion lines one by one.

[image]

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1

kaiwas, 2025-04-05:
In my video, the example with the selected edge was for the sake of simplicity of the demonstration.
In the "linked geometry" setting, you can select the vertices of the object. But in any case, this only complicates the process.

In general, there should be no difference between the subdivisions when we select an edge or we select vertices.

In addition, the process of applying dimensions is not intuitive.
If we first select the tool and then click on the edge, we get nothing. This is all very strange if compared with other programs.

https://github.com/user-attachments/assets/4c5405ff-6cec-4b81-9a95-48a7922c96d4

kaiwas, 2025-04-26:
In many other programs, snapping to vertices is done automatically. At the same time, you can create dimensions "in the air" (without snapping to vertices) 

https://github.com/user-attachments/assets/1e8bb736-e570-437f-8cc3-af926de4d74d

paullee0, 2026-01-08:
Cross-posting:
- https://forum.freecad.org/viewtopic.php?p=866367#p866367