FreeCAD/FreeCAD #20655: number of not-closed lines increases by just looking at a sketch

dirkx, 2025-04-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Every time you open this particular sketch; and then hit close (without making any changes) - the number of non-closed lines goes up by two. It has started off at 0 (with a pad).

[image]
[image]
[image]

### Full version info

```shell
OS: macOS 15.3.2
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Anything else?

Example file with the issue (but it seems to happe on any closed curve of which part is a spline):

https://pastebin.webweaving.org//try.IMSCrj.FCStd

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-04-05:
You have a problem point in the upper area of ​​the sketch. 

https://github.com/user-attachments/assets/1b002b7e-e3d3-4457-8aa5-6bd089f502bd

dirkx, 2025-04-05:
Thanks - but why does the number of non-closed curves go up by 2 every time I just open/close it ?

kaiwas, 2025-04-06:
> Thanks - but why does the number of non-closed curves go up by 2 every time I just open/close it ?

I don't know. But now the developers have a basis to understand what the problem is.

maxwxyz, 2025-04-06:
@PaddleStroke FYI

AjinkyaDahale, 2025-04-06:
@dirkx If you're talking of `Pad: wire is not closed (X times)` with the `X` increasing that's just the number of notifications. @maxwxyz, apart from this info, this issue appears to be a duplicate of a previous "wire not closed" issue.

closed 2025-04-06 by maxwxyz

AjinkyaDahale, 2025-04-06:
Just looked at the document shared. The wire is indeed not closed: There is a point-on-object from the bspline ends to the lines, but no point-on-object from the respective lines' ends on to the bspline. Alternatively these should be replaced by coincident. Here's one possible solution:

[trim_edge_issue_Post_Trim_in_1.0.FCStd.zip](https://github.com/user-attachments/files/19620455/trim_edge_issue_Post_Trim_in_1.0.FCStd.zip)