FreeCAD/FreeCAD #20656: failure to build with opencascade v7.9.0

ipatch, 2025-04-05:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

did a search in the issue tracker for opencascade and  did not see any mention of build / linking errors with opencascade v7.9.0.

i opened an issue with the homebrew-freecad repo below,

https://github.com/FreeCAD/homebrew-freecad/issues/672

it seems the arch package of opencascade hasn't been updated to v7.9.0 yet and as is still at version 7.8.1 also then gentoo package hasn't been updated either.

however there does seem to be a forum topic mentioning this issue, and fix is a simple one at that.

https://forum.freecad.org/viewtopic.php?style=2&t=94771

### Full version info

```shell
NA
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ipatch, 2025-04-05:
a simple PR with updated code in the forum topic should resolve this issue, and then a backport / cherry-pick of the PR / commit containing the fix should allow freecad v1.0 to build and link against opencascade version 7.9.0

ipatch, 2025-04-05:
the fix mentioned above in the forum topic,

```
$ cd src/FreeCAD
$ git diff cMake/FindOCC.cmake
diff --git a/cMake/FindOCC.cmake b/cMake/FindOCC.cmake
index c72066f43e..56d1593634 100644
--- a/cMake/FindOCC.cmake
+++ b/cMake/FindOCC.cmake
@@ -139,6 +139,7 @@ if(OCC_FOUND)
     TKPrim
     TKHLR
     TKFeat
+    TKExpress
   )
   set(OCC_OCAF_LIBRARIES
     TKBin
```

maxwxyz, 2025-04-06:
@pieterhijma FYI

ipatch, 2025-05-18:
just opened / submitted a PR to resolve this issue.

closed 2025-05-26 by yorikvanhavre