FreeCAD/FreeCAD #20661: Draft_Rotate error message when rotating subelements of objects that are not Wire or Line

semhustej, 2025-04-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open new file
- create a Draft_Rectangle
- select one edge of the Draft_Rectangle
- start Draft_Rotate
- enable `Modify subelements`
- pick any point as center of rotation
- pick any angle as a base angle
- pick any angle as a new angle
- error shows up

```
15:47:22  Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\Draft\draftutils\todo.py", line 165, in doTasks
    Gui.doCommand(string)
  File "<string>", line 1, in <module>
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\Draft\draftfunctions\rotate.py", line 92, in rotate
    return _rotate_subelements(selection, angle, center, axis, copy)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\Draft\draftfunctions\rotate.py", line 200, in _rotate_subelements
    rotate_edge(obj, edge_idx, angle, center, axis, global_place)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\Draft\draftfunctions\rotate.py", line 244, in rotate_edge
    rotate_vertex(obj, edge_idx, angle, center, axis, global_place)
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\Draft\draftfunctions\rotate.py", line 231, in rotate_vertex
    points = obj.Points
             ^^^^^^^^^^
AttributeError: 'FeaturePython' object has no attribute 'Points'

15:47:22  ToDo.doTasks, Unexpected error:
<class 'AttributeError'>
in ['Draft.rotate(selection, 21.026287257560142, center=FreeCAD.Vector(-57.717830657958984, 27.44811248779297, 0.0), axis=FreeCAD.Vector(0.0, 0.0, 1.0), copy=False, subelements=True)']

```

This error is not very clear for user to read.

See the behavior in video:

https://github.com/user-attachments/assets/d5a2f7b2-da9e-4f33-8c1b-e5a5dbce191c

**Proposed solution:**
Remove the error message and post an information for the user "Can not move subelements of this object. Uncheck the 'Modify subelements' option to modify this object".
More clear communication of the problem would enhance the FreeCAD user experience.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
```

### Subproject(s) affected?

Draft

### Anything else?

This happens with any non Line or Wire object.

The issue can be reproduce with FreeCAD 1.0

This issue can be observed with all these tools:

- Draft_Rotate
- Draft_Move
- BIM_Copy
- Draft_Scale

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-04-06:
I can confirm the error message 
```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
```

closed 2026-08-25 by Roy-043

semhustej, 2026-10-01:
Tested with the latest weekly. The issue is fixed, thank you.

```
OS: Debian GNU/Linux 13 (trixie) (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 26.3.0dev.20261001 (Git shallow) AppImage
Build date: 2026/10/01 12:05:00
Build type: Release
Branch: (HEAD detached at 99c5620c5)
Hash: 99c5620c5f739ca5c308849dda95ac468fec8b79
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: xcb/via Qt/via Qt
Navigation Style/Orbit Style/Rotation Mode: Revit/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/158.196/1
OpenGL version/vendor/renderer: 4.6 (Compatibility Profile) Mesa 25.0.7-2+deb13u1/Intel/Mesa Intel(R) HD Graphics 4600 (HSW GT2)

```