FreeCAD/FreeCAD #20662: Ubuntu 24 + new install of freecad 1.0 via snap or even appimage, addon manager not working

szucsnandor, 2025-04-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On a brand new Ubuntu 24 install I installed freecad using snap (and downloaded appimage too) and when open the addon manager the first time it stops on random downloads and hangs. No installed addons at all this is the first time a tried to open it. It does the same with appimage too. It might be a dependency incompatibilty but i dont get any error to check.  Also tried to delete the cache folder

[image]

.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: Hungarian/Hungary (hu_HU)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

### Subproject(s) affected?

None

### Anything else?

safe mode does the same

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

szucsnandor, 2025-04-07:
After starting a new document, adding a body, and then closing the document made the addon manager work. So it seems that a brand new install cannot use the addon manager before a document is created the first time.? Not sure if thats the case.

maxwxyz, 2025-04-07:
Does it also happen with the latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

szucsnandor, 2025-04-08:
> Does it also happen with the latest weekly: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

I downloaded that appimage but it won't even start. Any other appimage runs properly but not this weekly freecad build. Does it need any special linux lib?