FreeCAD/FreeCAD #20663: Datum Plane Attachment Mode: Face-to-Face [Feature Request]

SebDominguez, 2025-04-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When selecting 2 faces to create a datum it default to Inertia 2-3.

[image]

I think it would be really useful to have a Face-to-Face attachment mode where we can use a distance ratio between the two faces. Something similar to the Normal-to-Edge mode that uses the Map Path parameter to set the datum plane, for instance, at 0.5 (the middle of the edge). A Face-to-Face attachment mode set to 0.5 would result in a datum plane that is exactly between the two selected faces.

For instance, selecting both the top and bottom faces of the previous shape with a Face-to-Face attachment mode set to 0.5 should produce this datum plane:

[image]

### Full version info

```shell
OS: macOS 15.4
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.38
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Pendelf, 2025-04-06:
I confirm this behavior of the program

luzpaz, 2025-04-06:
@Pendelf confirmed on what version? (full about info pls)

SebDominguez, 2025-04-06:
> [@Pendelf](https://github.com/Pendelf) confirmed on what version? (full about info pls)

Same result in FreeCAD 1.1.0dev, with more errors in the report view, if that helps.
[image]

```
19:04:29  (qt.qpa.fonts) Populating font family aliases took 373 ms. Replace uses of missing font family "Latin Modern Roman" with one that exists to avoid this cost. 
19:06:23  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
19:06:23  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
19:06:23  AttachEngine3D::calculateAttachedPlacement:InertialCS: inertia tensor has axis of symmetry. Second and third axes of inertia are undefined.
```

```
OS: macOS Sequoia (15.4)
Architecture: x86_64
Version: 1.1.0dev.41138 (Git) Conda
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/Spain (en_ES)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/151/2
Installed mods: 
  * fasteners 0.5.38
  * lattice2 1.0.0
```