FreeCAD/FreeCAD #20664: FreeCAD freezes on other manifest namespaces

PhoneDroid, 2025-04-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

If an addon has a namespace that isn't either `''` or  
`https://wiki.FreeCAD.org/Package_metadata`, the  
addon manager errors out and causes FreeCAD to  
freeze up.

As the Dracula preference pack just adopted the new  
manifest schema, it now not only fails to load it's schema  
but also causes FreeCAD to freeze up.
( The new schema has a different namespace )

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.0dev.40971 (Git)
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/80.9812/1
Installed mods: 
  * SelectorToolbar
```

### Subproject(s) affected?

Addon Manager

### Anything else?

[image]

I propose the addon manager doesn't have a hardcoded  
namespace, instead it should use whatever namespace  
is present and parse with that.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PhoneDroid, 2025-04-06:
Note : The PR introducing the new schema / namespace to the Dracula addon has just been reverted - until this problem has been solved.

PhoneDroid, 2025-04-06:
Problem code located at https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/AddonManager/addonmanager_metadata.py#L275

PhoneDroid, 2025-04-06:
Fix
```py
@staticmethod
def _determine_namespace(root: ET.Element) -> str:

    pattern = r'(?P<Namespace>\{[\S]+?\})package'

    match = re.search(pattern,root.tag)

    if match :
        return match.group('Namespace')

    return ''
```

chennes, 2025-04-07:
The proposed code fails to raise an exception if the element isn't named "package" -- it should.

chennes, 2025-04-07:
How about:
```python
    @staticmethod
    def _determine_namespace(root: ET.Element) -> str:
        if root.tag.endswith("package"):
            return root.tag[:-len("package")]
        raise RuntimeError("No 'package' element found in metadata file")
```

PhoneDroid, 2025-04-08:
@chennes Sure. 

Btw why does it even raise an error in the first place? 
Shouldn't it just log & ignore invalid package manifests ..