FreeCAD/FreeCAD #20669: TNP- Simple model with a shapebinder changes face names without cause

JDMils, 2025-04-06:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This is a very simple model of a solid rectangle with thickness, adding 4 circles in each corner, padding and then shapebinding them in another body. After that, changes to the circle padding causes the faces of the circles to change names.
Example forum topic is here:
https://forum.freecad.org/viewtopic.php?p=820381

It seems the problem is with the Constrain Tangent function. Once I apply the constraint the shapebuilder fails. I created the whole model without the Constrain Tangent on all the circles and the model works fine. I then edit the Sketch with the circles and constrain tangent each one to the base outer edges and as soon as I action the last one (there are 8 constrain tangents in total for 4 circles), the shapebinder fails. All the faces of the padded circles are now renamed .

In the attached file, the faces do not rename, and thus always remain the same names, however the shapebinder fails when editing the circle padding and says "FaceX" is missing however it is still there and adding it back in with the name FaceX fixes the issue!?

I tested this thru all versions including 0.21.2 and it occurs in all of them.

I created a new model in FC and was able to recreate the issue. The cideo of the model creation is here:
https://youtu.be/aGSN4-Dzwsg
https://youtu.be/h_piZ5UiaBk

Sample file attached.
[ShapeBinder_JDMils.zip](https://github.com/user-attachments/files/19623927/ShapeBinder_JDMils.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41155 (Git) Conda
Build date: 2025/04/06 04:00:07
Build type: Release
Branch: main
Hash: e537278f1296211edb90bf1c938dfaebd2626f0d
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/Australia (en_AU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/106.57/1.5
Installed mods: 
  * A2plus 0.4.68
  * Assembly4 0.50.18
  * Curves 0.6.58
  * fasteners 0.5.38
  * Gridfinity 0.11.4
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

JDMils, 2025-04-07:
Even without the shapebinder, if I have the following model:
[image]

Notice the face names. I deleted the square shape on the top-left and all the face names have changed:
[image]

Zolko-123, 2025-04-10:
The TNP mitigation doesn't work. There are many very simple examples like yours that show it. But even though it doesn't work (reliably, sometimes it does but one can never know when it will work or fail) you pay the price in terms of lost performance to make the element map lookup.

JDMils, 2025-04-11:
I'm hoping my example will help the developers to fix the issue.

drwho495, 2025-05-18:
> I'm hoping my example will help the developers to fix the issue. 

Sadly no one is working on the top onaming issue right now

luzpaz, 2025-05-18:
Yes, indeed it was a huge blow to lose bgbsww[^1]

[^1]: https://blog.freecad.org/2024/10/08/rip-bradley-bgbsww-mclean/

drwho495, 2025-08-06:
hi there, i believe this is a known issue. (at least i know about it)
this could be fixable with complexfind
its also likely going to work the same in LS3