FreeCAD/FreeCAD #20670: Pocket/Pad - Two Dimension - Misleading Labels & Logic

Barniphife, 2025-04-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Pocket & Pad Task Panel options when using "Two-Dimension" mode include "Length", "2nd Dimension" and "Reverse" (among other things that I don't think are relevant here.)  "Length" is restricted to being a positive number, "2nd Dimension" is not.  "Reverse" seems to have the effect of negating the value entered for "Length".  However, in Two-Dimension mode, the "Length" value is actually the "starting position" of the pocket/pad and "2nd Dimension" is the "ending position".  The labels are misleading and the logic is confusing in cases where "Reverse" needs to be used.

As an example, consider the following:  I have a plate and I want to make 1 sketch on the front side of the plate and then use the hole tool to make a through-hole with a counterbore - all normal stuff that works as expected.  But then I want to use the same sketch to put a counterbore on the back side of the plate.  The hole tool can't be used for this second counterbore without creating a second sketch on the back side of the plate.  So, an alternative is to use the pocket tool in "Two-dimension" mode.  To setup the correct pocket parameters to get a pocket on the back side of the plate whereas the sketch is on the front-side of the plate, I can do the following:  1) Select Reverse, 2) set "Length" to the depth at which the pocket should begin.  I'm using a 1/2" plate and I want the pocket to be 1/8" deep as measured from the back face so I would choose "Length" to be 0.375" - having selected "reverse", this +0.375" value is treated as -0.375" which is what I want. 3) set the "2nd Dimension" to any negative value that is larger than the thickness of the plate - in my case, the plate is 1/2" thick so I can set "2nd Dimension" to -0.5" or any value more negative than that.

I realize that this example is trivial and there are other ways to do what I've described.  I'm raising the issue because the labels & logic could be better.  IMO, the appropriate labels could be "Dimension 1", "Dimension 2" or "Start Depth", "End Depth" where either value can be positive or negative.  Also, there is no need for the "Reverse" checkbox.

I appreciate the hard work of all the developers and I thank you in advance for considering this issue.  While I don't expect it to be a high priority, hopefully it can get worked in someday.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * A2plus 0.4.68
  * Assembly3 0.12.2
  * Assembly4 0.50.18
  * fasteners 0.5.37
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

hyarion, 2025-04-10:
The behavior will change with https://github.com/FreeCAD/FreeCAD/pull/20129

hyarion, 2025-04-10:
Regarding the names, I think Dimension 1 and 2 is better than Start/End Depth as depth is mostly meaningful when it comes to holes, but this can be used for any type of pocket/pad operation.

maxwxyz, 2025-04-13:
agreed

closed 2025-04-13 by maxwxyz