FreeCAD/FreeCAD #20676: BIM: IFC 'Convert to type' dialogue typo

semhustej, 2025-04-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- create new file
- create Arch_Wall
- convert the project to 'Strict IFC mode'
- right click the wall in the project browser and select IFC -> Convert to type
- the dialogue with %1 typo shows up

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41155 (Git) Conda
Build date: 2025/04/06 04:00:07
Build type: Release
Branch: main
Hash: e537278f1296211edb90bf1c938dfaebd2626f0d
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reinforcement
  * Reporting
  * Telemetry 1.0.0
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Gopal-001, 2025-04-08:
@maxwxyz can i take this up ?

maxwxyz, 2025-04-08:
@Gopal-001 sure, you can create a Pull Request

Gopal-001, 2025-04-08:
Sure @semhustej firstly i thought that we need to remove the %1 but after your suggestion i think there can a particular type replacing the %1. I'll look into it.

Gopal-001, 2025-04-08:
I saw in convert_to_type it's calling is_typable first where it's actually create a type and check if it's valid. I think that'd be the type which needs to be replaced from the placeholder rather than removing %1.

semhustej, 2025-04-08:
> I saw in convert_to_type it's calling is_typable first where it's actually create a type and check if it's valid. I think that'd be the type which needs to be replaced from the placeholder rather than removing %1.

That might be it.

I don't think it is that much important to show the type name to the user, so just removing the %1 is OK in my opinion.

semhustej, 2025-04-08:
@Gopal-001 one tip:

If you write "Fixes `link to this issue`" to the comment of the PR, this issue will automatically be closed when the PR is merged to main. See example here: https://github.com/FreeCAD/FreeCAD/pull/20400

semhustej, 2025-05-17:
Only now did I get to test the fix. It works nice with the dialogue showing the type information the object will be converted to.

Thanks for this contribution @Gopal-001 !

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41694 (Git)
Build date: 2025/05/13 12:46:54
Build type: Release
Branch: main
Hash: 5e624a4775c993a3d940f37876840c65ac0cc60c
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

Gopal-001, 2025-05-18:
Your welcome @semhustej , glad it worked 😊