FreeCAD/FreeCAD #20678: Thumbnails for file browsers not created on Linux

furgo16, 2025-04-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using a file browser (e.g. Nautilus on GNOME-based OSs), `.FCStd` files used to be shown as the thumbnail of the contents of the file. It currently does not seem to work, and the file is shown as a generic blank icon.

This is the script that generates the thumbnails:

https://github.com/FreeCAD/FreeCAD/blob/main/src/Tools/freecad-thumbnailer.in

It says in a comment:

> Register this thumbnailer
>    Adding a file /usr/share/thumbnailers/FreeCAD.thumbnailer with the following content:
>
>    [Thumbnailer Entry]
>    TryExec=freecad-thumbnailer
>    Exec=freecad-thumbnailer -s %s %i %o
>    MimeType=application/x-extension-fcstd;

However, taking the `freecad-daily` PPA's Debian package, I noticed:

1. The file to register a thumbnailer is shipped at `/usr/share/freecad-daily/share/thumbnailers/FreeCAD.thumbnailer` instead of `/usr/share/thumbnailers/FreeCAD.thumbnailer`
2. Listing the contents of `/usr/share/freecad-daily/share/thumbnailers/FreeCAD.thumbnailer` points to a `bin/freecad-thumbnailer` executable that is not installed by the Debian package
   ```
   [Thumbnailer Entry]
   TryExec=bin/freecad-thumbnailer
   Exec=bin/freecad-thumbnailer -s %s %i %o
   MimeType=application/x-extension-fcstd;
   ```

I'm not sure if point 1 makes a difference (can thumbnailers that are not in the `/usr/share/thumbnailers` location be registered?), but point 2 seems to be wrong.

### Full version info

```shell
1.1.0dev
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-04-07:
Some progress:

```
$ sudo update-mime-database -n -V /usr/local/share/mime/
Updating MIME database in /usr/local/share/mime...

Parsing source file /usr/local/share/mime/packages/org.freecad.FreeCAD.xml...
Wrote 2 strings at 2c - 6c
Wrote aliases at 6c - 70
Wrote parents at 70 - 74
Wrote literal globs at 74 - 78
Wrote suffix globs at 78 - d4
Wrote full globs at d4 - d8
Wrote magic at d8 - e4
Wrote namespace list at e4 - e8
Wrote icons list at e8 - ec
Wrote generic icons list at ec - f8
Wrote types list at f8 - 100
```

Running this made file browser thumbnails appear, and the rest are shown with the FreeCAD icon, albeit with the pre-1.0 icon.

I would have thought this command would be run by the .deb package's post-installation script, instead of having to be executed manually.

Also I don't know why there are two MIME database entries for FreeCAD:

```
/usr/local/share/mime/packages/org.freecad.FreeCAD.xml
/usr/share/freecad-daily/share/mime/packages/org.freecad.FreeCAD.xml
```

Finally, the `file` utility does not seem to recognize the FreeCAD MIME type and still thinks FreeCAD files are zip archives:

```
$ file --mime-type Barn\ door.FCStd 
Barn door.FCStd: application/zip
```

furgo16, 2025-04-08:
I don't know what installed `/usr/local/share/icons/hicolor/scalable/mimetypes/application-x-extension-fcstd.svg`, but it's the cause for the pre-1.0 icon showing up. My guess is that it was the old FreeCAD package from the Ubuntu archive (not the PPA), and it was left there.

If I manually replace the icon with the new one, and run:

```
$ sudo update-mime-database -n -V /usr/local/share/mime/
Updating MIME database in /usr/local/share/mime...

Parsing source file /usr/local/share/mime/packages/org.freecad.FreeCAD.xml...
Wrote 2 strings at 2c - 6c
Wrote aliases at 6c - 70
Wrote parents at 70 - 74
Wrote literal globs at 74 - 78
Wrote suffix globs at 78 - d4
Wrote full globs at d4 - d8
Wrote magic at d8 - e4
Wrote namespace list at e4 - e8
Wrote icons list at e8 - ec
Wrote generic icons list at ec - f8
Wrote types list at f8 - 100
```

Then the generic icon FreeCAD icon is shown for all .FCStd files. In summary:

1. Issue 1: MIME database needs to be updated manually
2. Issue 2: `freecad-thumbnailer` is not installed

Would it be possible that the original Ubuntu FreeCAD package did those steps, but the current package in the PPA does not?

furgo16, 2025-04-08:
On the upstream Debian package (`1.0.0+dfsg-6`):
- [x] FreeCAD thumbnailer registered
- [ ] FreeCAD thumbnailer installed
- [x] FreeCAD document MIME type registered

```
# locate thumbnailer
/usr/bin/gdk-pixbuf-thumbnailer
/usr/share/thumbnailers
/usr/share/glib-2.0/schemas/org.gnome.desktop.thumbnailers.gschema.xml
/usr/share/thumbnailers/FreeCAD.thumbnailer
/usr/share/thumbnailers/gdk-pixbuf-thumbnailer.thumbnailer
/usr/share/thumbnailers/librsvg.thumbnailer
```

```
# cat /usr/share/thumbnailers/FreeCAD.thumbnailer
[Thumbnailer Entry]
TryExec=freecad-thumbnailer
Exec=freecad-thumbnailer -s %s %i %o
MimeType=application/x-extension-fcstd;
```

```
# whereis freecad-thumbnailer
freecad-thumbnailer:
```

```
# locate org.freecad
/usr/share/applications/org.freecad.FreeCAD.desktop
/usr/share/icons/hicolor/16x16/apps/org.freecad.FreeCAD.png
/usr/share/icons/hicolor/32x32/apps/org.freecad.FreeCAD.png
/usr/share/icons/hicolor/48x48/apps/org.freecad.FreeCAD.png
/usr/share/icons/hicolor/64x64/apps/org.freecad.FreeCAD.png
/usr/share/icons/hicolor/scalable/apps/org.freecad.FreeCAD.svg
/usr/share/metainfo/org.freecad.FreeCAD.metainfo.xml
/usr/share/mime/packages/org.freecad.FreeCAD.xml
```

```
# cat /usr/share/mime/packages/org.freecad.FreeCAD.xml
<?xml version="1.0" encoding="UTF-8"?>
<mime-info xmlns='http://www.freedesktop.org/standards/shared-mime-info'>
  <mime-type type="application/x-extension-fcstd">
    <!-- <sub-class-of type="application/zip"/> -->
    <comment>FreeCAD document files</comment>
    <glob pattern="*.fcstd"/>
    <generic-icon name="application-x-extension-fcstd"/>
  </mime-type>
</mime-info>
```

luzpaz, 2025-07-21:
Is this still relevant ?

godiard, 2026-08-29:
Does not work on Fedora with FreeCAD installed from Flathub