FreeCAD/FreeCAD #20680: [UI/UX] Sketcher input units not showing properly

ErtugrulEge, 2025-04-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Sketcher input box are too narrow when showing units  and not responsive to the screen 

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41155 (Git) Conda
Build date: 2025/04/06 04:00:07
Build type: Release
Branch: main
Hash: e537278f1296211edb90bf1c938dfaebd2626f0d
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-07:
@kadet1090 @PaddleStroke FYI

kadet1090, 2025-04-07:
@alfrix could it be related to https://github.com/FreeCAD/FreeCAD/pull/20214 ?

furgo16, 2025-04-09:
Confirmed.

PR #20695 (2025-04-09): Gui: fix Quantity spinbox margin without an icon
margin refactoring aimed to fix #20680

margins are different depending on whether there is an icon or not

[image]

[image]
closed 2025-04-09 by kadet1090