FreeCAD/FreeCAD #20681: Weird constraint when creating a spline

Rigel-Alves, 2025-04-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi,

Not sure if this is the right place to post my request, but the forum is currently unavailable. I am trying to create a script that automates the generation of propeller blades in FreeCAD 1.0.0 for Linus. Everything goes well until the creation of the first (upper spline) in the code below; but when I try to add the lower spline, I get a weird redundant constraint:

```
import sys
import numpy as np

# NACA 2412
m = 0.02
p = 0.4
t = 0.12

def get_yt(x, t, definition):
    if definition == 'original':
        return 5*t*(0.2969*x**0.5 - 0.1260*x - 0.3516*x**2 + 0.2843*x**3 - 0.1015*x**4)

    elif definition == 'modified':
        return 5*t*(0.2969*x**0.5 - 0.1260*x - 0.3516*x**2 + 0.2843*x**3 - 0.1036*x**4)

def get_yc(x, m, p):
    return np.where((x >= 0) & (x < p), m/p**2 * (2*p*x - x**2), m/(1-p)**2 * ((1 - 2*p) + 2*p*x - x**2))
    # if (x >= 0) and (x < p):
    #     return m/p**2 * (2*p*x - x**2)
    # elif (x >= p) and (x <= 1):
    #     return m/(1-p)**2 * ((1 - 2*p) + 2*p*x - x**2)
    # else:
    #     print("invalid x")
    #     sys.exit()

def get_dycdx(x, m, p):
    return np.where((x >= 0) & (x < p), 2*m/p**2 * (p - x), 2*m/(1 - p)**2 * (p - x))
    # if x >= 0 and x < p:
    #     return 2*m/p**2*(p-x)
    # elif x >= p and x <= 1:
    #     return 2*m/(1-p)**2*(p-x)
    # else:
    #     print("invalid x")
    #     sys.exit()

def get_theta(x, m, p):
    dycdx = get_dycdx(x, m, p)
    return np.arctan(dycdx)

def get_coords(x, m, p, t, definition):
    yt = get_yt(x, t, definition)
    yc = get_yc(x, m, p)
    theta = get_theta(x, m, p)
    # x_upper, x_lower, y_upper, y_lower
    return [x - yt*np.sin(theta), x + yt*np.sin(theta), yc + yt*np.cos(theta), yc - yt*np.cos(theta)]

x = np.linspace(0.0, 1., 15)

definition = 'modified'

x_upper, x_lower, y_upper, y_lower = get_coords(x, m, p, t, definition)
print("x_upper =", x_upper)
print("y_upper =", y_upper)
print("x_lower =", x_lower)
print("y_lower =", y_lower)

# enforce y = 0 at chord end
y_upper[-1] = 0
y_lower[-1] = 0

constrained = True
document = 'Unnamed'

App.newDocument()
App.activeDocument().addObject('PartDesign::Body','Body')
App.ActiveDocument.getObject('Body').Label = 'Corpo'

chord_hub = 340 # mm
span = 1200 # mm
n_span = 11
spans = np.linspace(0.0, span, n_span) # mm
chords = np.linspace(chord_hub, 0.2*chord_hub, n_span) # mm
pitchs = np.linspace(0, 60, n_span) # degree
delta_x = np.linspace(0, 0.1*chord_hub, n_span) # mm
delta_y = np.linspace(0, 0.2*chord_hub, n_span) # mm

sketch_names = []
for span_id, span_height in enumerate(spans):
    sketch = 'Sketch' + str(span_id)
    sketch_names.append(sketch)

    App.getDocument('Unnamed').getObject('Body').newObject('Sketcher::SketchObject',sketch)
    App.getDocument('Unnamed').getObject(sketch).AttachmentSupport = (App.getDocument('Unnamed').getObject('XY_Plane'),[''])
    App.getDocument('Unnamed').getObject(sketch).MapMode = 'FlatFace'
    FreeCAD.getDocument('Unnamed').getObject(sketch).AttachmentOffset = App.Placement(App.Vector(delta_x[span_id],delta_y[span_id],span_height),App.Rotation(App.Vector(0,0,1),pitchs[span_id]))

    x_upper_chord = x_upper*chords[span_id]
    x_lower_chord = x_lower*chords[span_id]
    y_upper_chord = y_upper*chords[span_id]
    y_lower_chord = y_lower*chords[span_id]

    vectors_upper = []
    vectors_lower = []

    points_upper = []
    points_lower = []

    point_id = 0
    point_ids_upper = []
    point_ids_lower = []

    for i in range(len(x)):
        # suction side (upper points)
        vectors_upper.append(App.Vector(x_upper_chord[i], y_upper_chord[i], 0))
        points_upper.append(Part.Point(vectors_upper[i]))
        App.getDocument('Unnamed').getObject(sketch).addGeometry(points_upper[i], True) # the boolean flag is construction mode or not
        if constrained:
            App.getDocument('Unnamed').getObject(sketch).addConstraint(Sketcher.Constraint('DistanceX',point_id,1,x_upper_chord[i]))
            App.getDocument('Unnamed').getObject(sketch).addConstraint(Sketcher.Constraint('DistanceY',point_id,1,y_upper_chord[i]))
        point_ids_upper.append(point_id)
        point_id += 1

        # pressure side (lower points)
        vectors_lower.append(App.Vector(x_lower_chord[i], y_lower_chord[i], 0))
        points_lower.append(Part.Point(vectors_lower[i]))
        App.getDocument('Unnamed').getObject(sketch).addGeometry(points_lower[i], True) # the boolean flag is construction mode or not
        if constrained:
            App.getDocument('Unnamed').getObject(sketch).addConstraint(Sketcher.Constraint('DistanceX',point_id,1,x_lower_chord[i]))
            App.getDocument('Unnamed').getObject(sketch).addConstraint(Sketcher.Constraint('DistanceY',point_id,1,y_lower_chord[i]))
        point_ids_lower.append(point_id)
        point_id += 1

    #sys.exit()
    # suction side (upper spline)
    spline_upper_id = 2*len(x)

    _finalbsp_poles = []
    _finalbsp_knots = []
    _finalbsp_mults = []

    spline_upper = Part.BSplineCurve()
    spline_upper.interpolate(vectors_upper, PeriodicFlag=False)
    spline_upper.increaseDegree(3)

    _finalbsp_poles.extend(spline_upper.getPoles())
    _finalbsp_knots.extend(spline_upper.getKnots())
    _finalbsp_mults.extend(spline_upper.getMultiplicities())

    App.getDocument('Unnamed').getObject(sketch).addGeometry(Part.BSplineCurve(_finalbsp_poles,_finalbsp_mults,_finalbsp_knots,False,3,None,False),False)

    del(_finalbsp_poles)
    del(_finalbsp_knots)
    del(_finalbsp_mults)

    if constrained:
        conList = []
        for i in range(len(x)):
            conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',point_ids_upper[i],1,spline_upper_id,i))
        App.getDocument('Unnamed').getObject(sketch).addConstraint(conList)
        del conList
        App.getDocument('Unnamed').getObject(sketch).addConstraint(Sketcher.Constraint('Block',spline_upper_id))

    #sys.exit()
    # pressure side (lower spline)
    spline_lower_id = spline_upper_id + 1

    _finalbsp_poles = []
    _finalbsp_knots = []
    _finalbsp_mults = []

    spline_lower = Part.BSplineCurve()
    spline_lower.interpolate(vectors_lower, PeriodicFlag=False)
    spline_lower.increaseDegree(3)

    _finalbsp_poles.extend(spline_lower.getPoles())
    _finalbsp_knots.extend(spline_lower.getKnots())
    _finalbsp_mults.extend(spline_lower.getMultiplicities())

    App.getDocument('Unnamed').getObject(sketch).addGeometry(Part.BSplineCurve(_finalbsp_poles,_finalbsp_mults,_finalbsp_knots,False,3,None,False),False)

    del(_finalbsp_poles)
    del(_finalbsp_knots)
    del(_finalbsp_mults)

    if constrained:
        conList = []
        for i in range(len(x)):
            conList.append(Sketcher.Constraint('InternalAlignment:Sketcher::BSplineKnotPoint',point_ids_lower[i],1,spline_lower_id,i))
        App.getDocument('Unnamed').getObject(sketch).addConstraint(conList)
        del conList
        App.getDocument('Unnamed').getObject(sketch).addConstraint(Sketcher.Constraint('Block',spline_lower_id))

    sys.exit()
```

The code above can be run in FreeCAD using Crtl+F6. The weirdest thing is that the redundant constraint (9) is related to a point belonging to the first (upper) spline, which is successfully created.

Would it be a bug in FreeCAD? Could someone try to help me figure out why the code fails to generate a fully constrained sketch after the addition of the second (lower) spline?

Thanks a lot,

### Full version info

```shell
Version 1.0.0, 39109 (Git), Red Hat Enterprise Linux Server 7.9 (Maipo), x86_84
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-04-09:
@Rigel-Alves could you test this on the dev version[^1] and report back if you can reproduce the issue ?

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Rigel-Alves, 2025-04-13:
Thanks for your reply. Apologies for not providing the full About in my initial message; I was using my work PC. I have just tried now in my home PC with the default Ubuntu snap version, and the issue persists:

OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1248
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

I will now attempt with the dev version and let you know.

Rigel-Alves, 2025-04-13:
Just checked with:

OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.41232 (Git) Conda AppImage
Build date: 2025/04/11 14:31:02
Build type: Release
Branch: main
Hash: 1983052e381b5e5dbc78778186dc8f85a23163e5
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Portuguese/Brazil (pt_BR)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/102.359/1

The issue persists.

FEA-eng, 2026-02-13:
The sketches are still underconstrained.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260210 (Git shallow)
Build date: 2026/02/10 18:10:36
Build type: Release
Branch: (HEAD detached at ecbe21ca0)
Hash: ecbe21ca03a59723c90ce5a677ae7ccea06ebc12
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-08-16:
[This](https://github.com/FreeCAD/FreeCAD/issues/20681#issuecomment-4466193090) is still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 26.3.0dev.20260812 (Git shallow)
Build date: 2026/08/12 05:41:29
Build type: Release
Branch: (HEAD detached at dc1ef6e88)
Hash: dc1ef6e880d6f5cf5d5a6eb3749373141eb0deb8
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 610.88/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```