FreeCAD/FreeCAD #20682: BIM: 'IfcOpenShell update' always offers update in 1.1dev Windows

semhustej, 2025-04-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- start FreeCAD and enable BIM workbench
- check IfcOpenShell version:
```
>>> import ifcopenshell
>>> ifcopenshell.version
'0.8.1'
>>> ifcopenshell.version_core
'0.8.0'
```

- start 'Utils -> IfcOpenShell' update
- Update window shows:

[image]

- Select OK
- After upgrade seems to be run (no errors show up) check IfcOpenShell version

```
>>> # Gui.runCommand('IFC_UpdateIOS',0)
>>> ifcopenshell.version
'0.8.1'
>>> ifcopenshell.version_core
'0.8.0'
>>> 
```

- If 'Utils -> IfcOpenShell update' is run again, upgrade is offered again. B

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41155 (Git) Conda
Build date: 2025/04/06 04:00:07
Build type: Release
Branch: main
Hash: e537278f1296211edb90bf1c938dfaebd2626f0d
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reinforcement
  * Reporting
  * Telemetry 1.0.0
```

### Subproject(s) affected?

BIM

### Anything else?

I am not sure if it is related to this issue: https://github.com/FreeCAD/FreeCAD/issues/19821
Or this issue: https://github.com/FreeCAD/FreeCAD/issues/19821

Both of them seem to be solved though.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-04-07:
@semhustej the important part of the code is the [get_available_version](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/nativeifc/ifc_openshell.py#L155-L165) could you please change you local copy to:

```
    def get_avail_version(self):
        """Retrieves an available ifcopenshell version"""

        result = self.run_pip(["index", "versions", "ifcopenshell"])
        if result:
            print(result)
            if result.stdout and "versions" in result.stdout:
                result = result.stdout.split()
                result = result[result.index("versions:")+1:]
                result = [r.strip(",") for r in result]
                print(result[0])
                return result[0]  # we return the biggest
        return None
```

and report back the output, I'm expecting something like:

```
19:22:16  CompletedProcess(args=['/usr/bin/python3', '-m', 'pip', '--disable-pip-version-check', 'index', 'versions', 'ifcopenshell'], returncode=0, stdout='ifcopenshell (0.8.1.post1)\nAvailable versions: 0.8.1.post1, 0.8.1, 0.8.0, 0.7.10, 0.7.0.240627, 0.7.0.240521, 0.7.0.240518, 0.7.0.240423, 0.7.0.240418, 0.7.0.240406\n', stderr='WARNING: pip index is currently an experimental command. It may be removed/changed in a future release without prior warning.\n')
19:22:16  0.8.1.post1
```

which correctly stated that my copy is already up-to-date using a three week old build:

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40759 (Git)
Build date: 2025/03/17 05:00:02
Build type: Release
Branch: (HEAD detached at fd4fd6ccf7)
Hash: fd4fd6ccf7e0316742ebd4235ab6251059d55275
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, IfcOpenShell 0.8.1.post1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/93.1131/1
Installed mods: 
  * FreeCAD-Ribbon 1.8.1.2 (Disabled)
  * btl 0.9.9
  * Reinforcement
  * FreeCAD-themes 2025.1.7
  * FOSS3DLiveLink.zip
  * DynamicData 2.73.0
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.30.1
  * ChangeAppFont
  * Quetzal 1.3.3
  * fasteners 0.5.38
  * appimage_updater
  * freecad.gears 1.3.0
  * Curves 0.6.60
  * FOSS3DLiveLink
  * A2plus 0.4.68
  * ThreadProfile 1.93.0

```

semhustej, 2025-04-08:
> [@semhustej](https://github.com/semhustej) the important part of the code is the [get_avail_version](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/nativeifc/ifc_openshell.py#L155-L165) could you please change you local copy to:
> 

Thanks for looking into the issue. 
I ran with the adjusted code. This is the output in the report view - it seems it is what you expected. The last timestamp is red with no information:

```
15:11:18  CompletedProcess(args=['C:\\Users\\semhu\\Documents\\tyo\\BF\\FreeCAD\\FC_daily\\FreeCAD_weekly-builds-41155-conda-Windows-x86_64-py311\\bin\\python.exe', '-m', 'pip', '--disable-pip-version-check', 'index', 'versions', 'ifcopenshell'], returncode=0, stdout='ifcopenshell (0.8.1.post1)\nAvailable versions: 0.8.1.post1, 0.8.1, 0.8.0, 0.7.10, 0.7.0.240627, 0.7.0.240521, 0.7.0.240518, 0.7.0.240423, 0.7.0.240418, 0.7.0.240406\n', stderr='WARNING: pip index is currently an experimental command. It may be removed/changed in a future release without prior warning.\n')
15:11:18  0.8.1.post1
15:11:37 
```

semhustej, 2025-04-09:
Interesting. I am not using BitDefender or any other anti-virus non-MS software. I use only built-in Windows virus and threat protection.

@ErtugrulEge are you able to reproduce this issue on your Windows machine? I checked this issue on one machine.

Syres916, 2025-04-09:
@semhustej did you try the trick of deleting the `ifcopenshell` folder under `bin\Lib\site-packages` which on starting FreeCAD and running `Utils -> IfcOpenShell` triggers a new install into `%appdata%\Roaming\FreeCAD\AdditionalPythonPackages\py311` ?

semhustej, 2025-04-10:
I did these steps:

- delete `ifcopenshell` directory from `bin\Lib\site-packages`
- start FreeCAD
- start `Utils -> IfcOpenShell update`
- I am offered version `0.8.1.post1`, so I click `OK`
- report view shows red timestamp with no information
- check IOS version:
```
>>> ifcopenshell.version
'0.8.0'
>>> ifcopenshell.version_core
'0.8.0'
```

- When running `Utils -> IfcOpenShell update` I am offered version `0.8.1.post1`

After this, `ifcopenshell` directory is not in `bin\Lib\site-packages` folder.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Conda
Build date: 2025/04/09 18:28:54
Build type: Release
Branch: main
Hash: de124a72b56515a34f8ed627a768b54695698fc8
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reinforcement
  * Reporting
  * Telemetry 1.0.0

```

ErtugrulEge, 2025-04-10:
> Interesting. I am not using BitDefender or any other anti-virus non-MS software. I use only built-in Windows virus and threat protection.
> 
> [@ErtugrulEge](https://github.com/ErtugrulEge) are you able to reproduce this issue on your Windows machine? I checked this issue on one machine.

I can confirm 

[image]
```

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41155 (Git) Conda
Build date: 2025/04/06 04:00:07
Build type: Release
Branch: main
Hash: e537278f1296211edb90bf1c938dfaebd2626f0d
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
```

semhustej, 2025-08-21:
Note: I did a test with latest weekly and the issue still prevails. It offers upgrade to IfcOpenShell 0.8.3.post2, but the update does not happen.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.2

```

Roy-043, 2025-09-20:
This may be due to the order of the Python `sys.path`. On my Windows machine `'<install_folder>\\bin\\Lib\\site-packages'` appears before `'C:\\Users\\<user_name>\\AppData\\Roaming\\FreeCAD\\AdditionalPythonPackages\\py311'`. The update tool installs files in the latter folder, but when searching for the files the first folder has priority I think.

You can enter `sys.path` in the Python Console.

semhustej, 2025-09-21:
> This may be due to the order of the Python `sys.path`. On my Windows machine `'<install_folder>\\bin\\Lib\\site-packages'` appears before `'C:\\Users\\<user_name>\\AppData\\Roaming\\FreeCAD\\AdditionalPythonPackages\\py311'`. The update tool installs files in the latter folder, but when searching for and loading the files the first folder has priority I think.
> 
> You can enter `sys.path` in the Python Console.

You are right. `sys.path` gives this output:

Positioned first:  ` 'C:\\Users\\semhu\\Documents\\tyo\\BF\\FreeCAD\\FC_daily\\FreeCAD_weekly-2025.09.17-Windows-x86_64-py311\\bin\\Lib\\site-packages'`

Positioned second: `  'C:\\Users\\semhu\\AppData\\Roaming\\FreeCAD\\v1-1\\AdditionalPythonPackages\\py311'`

chennes, 2025-09-26:
`AdditionalPythonPackages ` was not intended to override FreeCAD-included packages, but that's almost coincidental, because from the Addon Manager's perspective no FreeCAD-included package should ever get installed there (the Addon Manager checks to see if the package can be imported before it allows installation). However, ifcOpenShell wasn't always included with FreeCAD, so maybe this installation predates its inclusion.

Roy-043, 2025-09-27:
Thanks. I'll try to find/create a suitable folder that is higher in the `sys.path` then.

semhustej, 2025-09-29:
I tested the library using the 1.1dev version. I successfully updated IOS from 0.8.2 to 0.8.3.post2. It works OK.
I did not test with lowering the version number.

```
>>> import ifcopenshell
>>> ifcopenshell.version
'0.8.2'
>>> # Gui.runCommand('IFC_UpdateIOS',0)
>>> ifcopenshell.version
'0.8.3.post2'
>>> ifcopenshell.__file__
'C:\\Users\\semhu\\AppData\\Roaming\\FreeCAD\\v1-1\\Mod\\UpdatedPythonPackages\\ifcopenshell\\__init__.py'
>>> 
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.3.post2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.2
  * UpdatedPythonPackages

```

ErtugrulEge, 2026-04-08:
Still related 

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260407 (Git shallow)
Build date: 2026/04/07 22:56:13
Build type: Release
Branch: (HEAD detached at 6db886fe0)
Hash: 6db886fe0e79bf060ffeaa50198da25702a17e31
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods:
  * Curves workbench 0.6.70

```

luzpaz, 2026-09-22:
> I don't think there would be a big harm in not exposing this option to user. If it is complicated to implement, I think there is lot of low hanging fruit with regards to IFC implementation in FreeCAD that would be better to focus on than this one.

So should this option be disabled for now ? 

JFYI, IfcOpenShell was just updated in https://github.com/FreeCAD/FreeCAD/pull/32431

semhustej, 2026-09-22:
> 
> So should this option be disabled for now ?
> 

If it is not possible (or too complicated - I see there are different issues with IOS packaging on various platforms or even installation options) to make it work, I think it would be better to hide the tool from the user.

@Roy-043 what is your opinion?