FreeCAD/FreeCAD #20692: Mouse cursor shouldn't move during 3D navigation

jkprg, 2025-04-07:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When user pans or rotates in 3D view and the mouse cursor gets over panes (tree view, property view etc.) the navigation is paused (the 3D view doesn't get mouse events). The effect is more painful with transparent panes...

I would suggest to freeze somehow mouse cursor in place with appropriate icon (panning/rotating icon) while still generating correct motion events although I'm not sure if the selected GUI toolkit allows such a behavior.

### Full version info

```shell
OS: macOS Sequoia (15.3)
Architecture: arm64
Version: 1.1.0dev.41138 (Git) Conda
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/Czechia (en_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * Plot 2024.11.26
  * FreeCAD-themes 2024.7.24
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * CfdOF 1.29.4
  * Curves 0.6.59
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-08:
@Rexbas is this possible?

ErtugrulEge, 2025-04-08:
I can't confirm this. My navigation style is Blender and with middle mouse I can easily 3d navigate over tree view or property view.Please share some images or better a  video.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.41155 (Git) Conda
Build date: 2025/04/06 04:00:07
Build type: Release
Branch: main
Hash: e537278f1296211edb90bf1c938dfaebd2626f0d
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/windowsvista
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

Rexbas, 2025-04-09:
@jkprg, I think I can reproduce this but only with the Touchpad style. Are you using the Touchpad style or another?

jkprg, 2025-04-09:
> [@jkprg](https://github.com/jkprg), I think I can reproduce this but only with the Touchpad style. Are you using the Touchpad style or another?

Wow. You are right. I didn't notice that. The problem exists only in Touchpad navigation style so far. OpenSCAD works as fine...

ErtugrulEge, 2025-04-10:
`OS: macOS Sequoia (15.3)`  maybe MacOS related

luzpaz, 2025-11-09:
@Rexbas any possibility you can fix for 1.1 ?

FEA-eng, 2026-09-30:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```