FreeCAD/FreeCAD #20699: Trying to model knurling pattern on cylinder and Polar and Mirror features have problems

Jaron123, 2025-04-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Before I talk about my problem I would like to say I am incredibly thankfull to all the developers for making this piece of software. I have tried it many times over the years and always ran up against problems that made it not possible for me to use and recently with the 1.0 update I gave it another shot. And almost immediately ran up to a feature that is impossible to model in FreeCad it seems.
I wanted to model the back of an audio connector that has some knurling on it. Making this knurling seems to be impossible, I made a little test-case and attached it to this issue. 
My hope is that this account can help development focus on what an average person trying to model with FreeCad runs up against which could make it not that usable for Cad design work..
The workflow I wanted to follow was this, Make a cylinder, apply helix cut, circular pattern the helix (polar pattern), mirror the circular pattern.
I already figured out that the Circular and mirror pattern can only be applied together when using a Multi-transform feature.
I made an example project that shows the problem, make a cylinder, apply a helix cut, and FreeCad doesn’t like circular pattern with an instance more than 3… (It looks like it starts creating new bodies or something.. Or the planes get detached when cutting, I’m not sure.. It gets even worse when we try to apply a mirror to those circular instances..
This example case could use a circular pattern with an instance of about 10 but knurling often has to go to 80 circular cut instances or so. Which causes even more problems and the Multi-transform refuses to apply because it’s creating multiple bodies or it freezes the program

[Test_Model_Knurling.zip](https://github.com/user-attachments/files/19647804/Test_Model_Knurling.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Netherlands (en_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-04-09:
Unfortunately, this is such a known issue that someone turned it into a meme:
 
[image]

ref: https://forum.freecad.org/viewtopic.php?style=10&p=480292

Roy-043, 2025-04-09:
Try changing the Refine property of the pad and the helix to false.

trevarj, 2025-05-06:
> Try changing the Refine property of the pad and the helix to false.

Came here having the same issue and this worked perfectly and rendered quickly for me.

trevarj, 2025-05-18:
I suppose I should note that when doing a very dense knurling, with many occurrences, FreeCAD sucks up all RAM and finally halts the PC (then gets killed by OS). Would be nice to know why or have a way to apply knurling without rendering, kind of like with threaded holes, then finally render it.

FEA-eng, 2025-12-14:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251209 (Git shallow)
Build date: 2025/12/09 13:53:17
Build type: Release
Branch: grafted,grafted
Hash: 5ad6a1ba58c665c23f1d217c23e74379d8c1a5a4
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 621.01/NVIDIA Corporation/NVIDIA GeForce RTX 5080/PCIe/SSE2
```