FreeCAD/FreeCAD #20703: [Feature] Selected item locating effect

Beep6581, 2025-04-08:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Please add a feature which somehow brings attention to where the selected item is in the preview.

Anything beyond the simplest of designs will consist of overlapping geometries, points and constraint lines. Furthermore, even non-overlapping features can be difficult to locate, owing to their size. When I select something in the treeview, or when I select a constraint or element in Sketcher, it can be difficult and even impossible to find it in the main preview. It would be a great benefit if there was a feature which somehow brings attention to the selected item, e.g. by temporarily showing a large crosshair centered on the center-of-mass of the selected object.

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41138 (Git) Conda AppImage
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/141.951/1
Installed mods: 
  * fasteners 0.5.37
  * A2plus 0.4.68
  * ThreadProfile 1.92.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-10:
In other words, the selection effect should mimic the hovering mouse effect

Beep6581, 2025-11-20:
Confirmed that this works in the main view, but there is room for improvement in Sketcher.

In Sketcher, when I select an element in the preview, that element is selected in the "Elements" list, but FreeCAD does not bring it into view. Some time ago I wrote a patch which was merged into the codebase for the Constraint list scrolling the selected constraints into view - I will do the same for Elements.

[image]

There are two more issues with selecting elements.
1. It is very difficult to find the selected point. Note that, in the screenshot, a point is selected in the Elements panel, but its next to impossible to find it in the preview. Hopefully this is handled by https://github.com/FreeCAD/FPA-grant-proposals/issues/65
2. If I want to change the color of the selected point, I would expect to go to Preferences > Sketcher > Appearance. But alas, no. The color of the selected point is defined by Preferences > General > Selection > Viewport Selection Behavior > Enable Selection, and it affects a lot more than just Sketcher selections. This is partially covered by #22529 , but the color of un/selected points should have its own dedicated setting under Preferences > Sketcher > Appearance.