FreeCAD/FreeCAD #20714: [PD] The object and sketch are not displayed in the tree view until they are selected in the 3D window.

kaiwas, 2025-04-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Yesterday my colleague sent me a file with a problem.
When opening the file, we see the body with the origin in the tree view, but we do not see the PAD and sketch.
In order for the PAD and sketch to appear, you need to select them in the 3D window.

The problem is not solved by resaving the file. The next time you open it, the situation will repeat.

I do not know what exactly the problem could be related to. The console and report do not show errors.
I know that the file was originally created in version 1.0 without any specific settings.

Video demonstration.

https://github.com/user-attachments/assets/428c13c5-71fe-4455-beeb-994ebe01565a

The file is attached. Remove the ZIP extension

[18650 Case.FCStd.zip](https://github.com/user-attachments/files/19660906/18650.Case.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41138 (Git) Conda AppImage
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Test
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.60
  * freecad.gears 1.3.0
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-04-09:
RMB on the tree, _Show items hidden in the tree view_

When shown you can _Toggle the visibility in tree view_  for each object independently and switch _Show items …_ back

kaiwas, 2025-04-09:
> RMB on the tree, _Show items hidden in the tree view_
> 
> When shown you can _Toggle the visibility in tree view_ for each object independently and switch _Show items …_ back

Thanks! It helped.
But immediately the idea arose that there is no information (identifier) ​​that there are hidden objects in the tree view.

Maybe this issue should be "corrected" for this request?

kaiwas, 2025-04-09:
> hidden in the tree view

Moreover, by hiding unnecessary sketches, the problem of summing up the displacements will be solved.
It seems that I have a good idea. I will think it over and make a feature request.
Once again, many thanks! You solved a lot of my problems with your hint.

kaiwas, 2025-04-09:
It is also unclear why the object is shown in the tree if it is selected. In my opinion, this should not happen.

In general... a series of ideas are needed to improve this algorithm. I will make a request with examples.

kaiwas, 2025-04-09:
In order not to create many issues I created a discussion on the topic of hiding objects. I will be glad if you take part. I think this issue is very important. It can greatly improve the convenience of work.

https://github.com/FreeCAD/FreeCAD/discussions/20721

closed 2025-04-12 by maxwxyz