FreeCAD/FreeCAD #20716: CAM changing ExtraOffset is GUI dlg fails to change value.

J-Dunn, 2025-04-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Setting " Extra Offset" in the Helix dialogue has no effect. That change is not reflected in the object property "Offset Extra".  Editing the value in the properties table works as expected. 

The property value is transmitted to the dlg but any changes there are ignored on clicking "Apply" or closing with "OK". 

This seems true with several AppImages or 1.0 and 1.1 as well as current master.  It seems likely it never worked. 

https://forum.freecad.org/viewtopic.php?t=96160

### Full version info

```shell
OS: Fedora Linux 40 (Forty) (LXQt/lxqt/xcb)
Architecture: x86_64
Version: 1.1.0dev.41073 (Git)
Build date: 2025/03/29 23:43:40
Build type: Release
Branch: main
Hash: e7156aa32608e7d06eefb2fe7fd0dd8d1aa7f138
Python 3.12.9, Qt 6.8.2, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/98.4815/1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tarman3, 2025-04-09:
[image]

<details>
  <summary>FreeCAD 1.1.0dev.41127 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>
```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41127 (Git) Conda AppImage
Build date: 2025/04/01 15:11:50
Build type: Release
Branch: main
Hash: db08735e106561f3a11cf08e02fbf5d9e8fbf6b2
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/118.293/1
Installed mods: CurvedShapes 1.0.13; lattice2 1.0.0; Silk 0.1.5; freecad.gears 1.3.0; btl 0.9.9 (Disabled); fasteners 0.5.38; MeshRemodel 1.10.37; Curves 0.6.60; sheetmetal 0.7.22; Defeaturing 1.2.3

```
</details>

J-Dunn, 2025-04-09:
Interesting.   Could you dump your version information to compare to what I posted? 

These AppImages are not totally system independent.  I have not looked at what the include but I tried the most recent 0.21 AppImage and it did a core dump off the bat. 

This implies some dependence on installed system libraries, which could be the reason for these differences.

tarman3, 2025-04-09:
[cam_helix_test.zip](https://github.com/user-attachments/files/19664861/cam_helix_test.zip)

Checked with

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.0, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.36; freecad.gears 1.3.0; fasteners 0.5.36; sheetmetal 0.7.5; btl 0.9.9; Silk 0.1.5; Curves 0.6.54; Defeaturing 1.2.3

```

</details>

<details>
  <summary>FreeCAD 1.1.0dev.41127 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>
```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41127 (Git) Conda AppImage
Build date: 2025/04/01 15:11:50
Build type: Release
Branch: main
Hash: db08735e106561f3a11cf08e02fbf5d9e8fbf6b2
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/118.293/1
Installed mods: CurvedShapes 1.0.13; lattice2 1.0.0; Silk 0.1.5; freecad.gears 1.3.0; btl 0.9.9 (Disabled); fasteners 0.5.38; MeshRemodel 1.10.37; Curves 0.6.60; sheetmetal 0.7.22; Defeaturing 1.2.3

```
</details>

tarman3, 2025-09-02:
Can be fixed by #21971

tarman3, 2026-01-15:
Can not reproduce in 1.2.0dev.20260106 (Git shallow) AppImage