FreeCAD/FreeCAD #20718: [Part] Relevance Part B-spline weight assignment workaround

pieterhijma, 2025-04-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Part has a workaround for a potential OCCT problem. I'm wondering if this is still relevant for newer OCCT versions, so I would like to track this issue in the OCCT Liaison project.

### Full version info

```shell
OS: Arch Linux (xcb)
Architecture: x86_64
Version: 1.1.0dev.41211 (Git)
Build date: 2025/04/08 20:56:21
Build type: Release
Branch: main
Hash: ce0080cbae450fd62aadd41292cfd6e17723bf3d
Python 3.13.2, Qt 6.9.0, Coin 4.0.3, Vtk , OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
```

### Subproject(s) affected?

Part

### Anything else?

The commit that introduced the workaround: 324f6d3e260dd5fd114477db09d3c4557d8e1992

The log message of the commit:

Part: Geometry - BSpline - fix weight assignment
    
================================================
    
If during assignment of weights (during the for loop iteratively setting the poles) all weights
become (temporarily) equal even though weights does not have equal values
OCCT will convert all the weights (the already assigned and those not yet assigned)
to 1.0 (nonrational b-splines have 1.0 weights). This may lead to the assignment of wrong
of weight values.
    
The work-around is to temporarily set the last weight to be assigned to a value different from
the current value and the to-be-assigned value for the weight at position last-to-be-assign but one.
    
Fixes:
https://forum.freecadweb.org/viewtopic.php?p=478701#p478702

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

pieterhijma, 2025-04-09:
@AjinkyaDahale is willing to take a look. Thanks in advance!

adrianinsaval, 2025-04-22:
unrelated to that problem but you might also want to look into the relevance of  https://github.com/FreeCAD/FreeCAD/blob/efbfa34aa19089222caa2c7032806c44f1360492/src/Mod/Part/App/BRepOffsetAPI_MakeOffsetFix.cpp

stefano2734, 2026-09-12:
Last messages are here more than one year old.
Perhaps here an improvement with 26.3 and occt 7.9.3 and new 8.0.1.