FreeCAD/FreeCAD #20722: Placement indicator display artifacts for link objects

kaiwas, 2025-04-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When we select a link object in the tree view, the letter designations of the axes get an outline.

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.41138 (Git) Conda AppImage
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Test
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.60
  * freecad.gears 1.3.0
  * sheetmetal 0.7.22
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-05-17:
Any relationship to... ?
* #19894

kaiwas, 2025-05-17:
> Any relationship to... ?

No. The problem indicated in the link was identified earlier than mine.