FreeCAD/FreeCAD #20724: Core: values should be displayed at a specific precision and not with a specific number of decimal

FlachyJoe, 2025-04-09:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When using small objects in a large project, value display issues may occur. Automatically switching values ​​to the best metric multiple results in a loss of accuracy.

As a workaround you can type the requested value in expression
[image]

If we could set an absolute precision, e.g. 0.1mm, even the large values will be displayed correctly.

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.41188 +11 (Git)
Build date: 2025/04/08 06:33:25
Build type: Debug
Branch: LockedProperty
Hash: 7be74d5cc1b406e9f8e69a80e2e79f1aca34c02d
Python 3.13.2, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.4394/1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-13:
@FreeCAD/design-working-group @FreeCAD/cad-working-group FYI

pierreporte, 2025-05-01:
I agree that, whenever possible, the exact entered value should be displayed, but what happens if there is not enough room? It should be a rare occurrence though, but in such cases, using the ≈ sign proposed for #20618 would indicate that the displayed value is not the exact one, which would be visible only when editing.