FreeCAD/FreeCAD #20726: SVG: Imported text not rendered properly or at all

psommerfeld, 2025-04-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Compare the Draft and TechDraw screenshot to the Inkspace screenshot.
In Draft:
-  it seems text objects in an SVG don't appear at all, even though Tree View acknowledges them.
In TechDraw:
- some text appear, others are missing.
- specifically, in the test, "abc" string appears, however it is misplaced.
- "P" and "S" seem to be rendered correctly.
- "Q", which Inkscape somehow stores differently in the .svg, is missing altogether.

[draft svg deformed.FCStd.zip](https://github.com/user-attachments/files/19676517/draft.svg.deformed.FCStd.zip)
[techdraw svg deformed.FCStd.zip](https://github.com/user-attachments/files/19676518/techdraw.svg.deformed.FCStd.zip)

### Full version info

```shell
OS: macOS Sequoia (15.3)
Architecture: arm64
Version: 1.1.0dev.41138 (Git) Conda
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```

### Subproject(s) affected?

Draft

### Anything else?

Compare to Google Chrome, which renders the SVG perfectly, showing that Inkscape is not at fault here.

[image]
[image]
[image]
[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

phaseloop, 2025-04-11:
May be interesting to @aprospero

aprospero, 2025-04-12:
Could you provide the svg?

psommerfeld, 2025-04-12:
Attached

> Could you provide the svg?

[image]

psommerfeld, 2025-04-12:
Can't delete my previous comment for some reason but I didn't realize I hadn't clipped the canvas in Inkscape to minimum size and that it would show the SVG inline. Here's a more sane version.

[image]

aprospero, 2025-04-14:
Thanks for the svg!
Unfortunately, there is not much I can say about this. In reflection of your Issue I rephrased my PR from 'importSVG Overhaul' to 'importSVG - path generation Overhaul'. I solely concentrate on path generation and surroundings.

Out of interest: what is your typical workflow when you import svg text elements? How do your post-process them, how do you use them in general?

aprospero, 2025-04-14:
I forgot to mention: I can't see the imported text elements in the model view until I set the View Parameter 'Display Mode' from 'Screen' to 'World'. 

I haven't checked why it is set in the first place, though.

luzpaz, 2025-09-26:
bumping for visibility

FEA-eng, 2025-11-26:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251126 (Git shallow)
Build date: 2025/11/26 00:30:17
Build type: Release
Branch: (HEAD detached at de230cf)
Hash: de230cfaa75433441366ad09a4d5fda828e7bfdb
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-02-25:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260224 (Git shallow)
Build date: 2026/02/24 18:54:53
Build type: Release
Branch: (HEAD detached at 34ae16cd0)
Hash: 34ae16cd01b179eb9e1801591276dbc5a38669b5
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```