FreeCAD/FreeCAD #20733: Deleting objects causes crash

psommerfeld, 2025-04-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As shown in video:
1. Open attached project
2. Delete the last 3 objects (named: orig window, door, window sk) in the tree at one time.
3. In the dialog that pops up, select 'Yes' to continue.

[delete causes crash.FCStd.zip](https://github.com/user-attachments/files/19684838/delete.causes.crash.FCStd.zip)

FreeCAD will instantly crash

### Full version info

```shell
OS: macOS Sequoia (15.3)
Architecture: arm64
Version: 1.1.0dev.41138 (Git) Conda
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/Canada (en_CA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
```

### Subproject(s) affected?

PartDesign

### Anything else?

Most of the time, a crash report is not presented, but here is one that I captured:
`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [35892]
Path:                  /Applications/FreeCAD dev.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-04-10 08:04:53.5039 -0400
OS Version:            macOS 15.3.1 (24D70)
Report Version:        12
Anonymous UUID:        56FBAF43-D499-48FD-388A-022E72B815D2

Sleep/Wake UUID:       E72BFBC9-9807-4053-9011-920C067CA965

Time Awake Since Boot: 120000 seconds
Time Since Wake:       2907 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [35892]
[... 483 lines cut ...]
__INFO_FILTER                         8        1 
__LINKEDIT                       668.0M      165 
__OBJC_RW                         2374K        1 
__TEXT                             1.2G     1091 
__TPRO_CONST                       272K        2 
dyld private memory                160K        2 
mapped file                      546.5M       74 
owned unmapped memory             1552K        1 
page table in kernel              1458K        1 
shared memory                     1024K       25 
===========                     =======  ======= 
TOTAL                              5.3G     5951 
TOTAL, minus reserved VM space     4.9G     5951 

-----------
Full Report
-----------
..
  "usedImages" : [
  {...}
`

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-04-10:
The backtrace is similar to #20498. In both cases an orphan object shows up and `Gui::DocumentObjectItem::testStatus(bool, QIcon&, QIcon&)` does not expect it.

Syres916, 2025-07-23:
@maxwxyz is a month of no response enough for closure?