FreeCAD/FreeCAD #20736: File content starts folded

leoheck, 2025-04-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Files are opening folded

[image]

Can we store this inside the files so we can keep them opening unfolded?
It does not make sense to have them folded when the Model view is set to show a single File.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.41213 (Git) Conda
Build date: 2025/04/09 15:48:15
Build type: Release
Branch: main
Hash: f258a2639cd8acd45cb101b7f34e90e603d91696
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/145.143/1
Installed mods: 
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.73.0
  * Curves 0.6.60
  * InventorLoader 1.5.1
  * ose-piping
  * OpenTheme 2024.9.1
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.1.3
  * lattice2 1.0.0
  * kicadStepUpMod 11.6.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-13:
@FreeCAD/design-working-group FYI

pierreporte, 2025-04-13:
I agree that it’s not great. IMO it should open the first children of the document object, when we have the new file format I suggested, the root assembly, part, or drawing would have their first children directly visible. Just making visible the first children (without unfolding them) wouldn’t look great in this case.

Syres916, 2025-04-15:
In 0.21.x the tree opens exactly as the user saved the file, surely that is what a user expects?