FreeCAD/FreeCAD #20740: FreeCAD Dark/Light: property values invisible when an expression value is selected on the Property Editor

furgo16, 2025-04-10:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When selecting the value of a property that contains an expression in the Property Editor, the cell is shown as blank. The effect is that the value is no longer visible. Probably it becomes white text on white background (or black on black):

FreeCAD Light
[image]

FreeCAD Dark
[image]

FreeCAD Classic
[image]

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.41214 (Git) Snap 1380
Build date: 2025/04/09 18:28:54
Build type: Release
Branch: main
Hash: de124a72b56515a34f8ed627a768b54695698fc8
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 9.1.0, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * Assembly4 0.50.17
  * Launcher
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-04-11:
@MisterMakerNL FYI

MisterMakerNL, 2025-04-11:
That is annoying! I'll have look asap!

MisterMakerNL, 2025-04-11:
I fixed it in my pr, I just don't know how.... :)
[image]
https://github.com/FreeCAD/FreeCAD/pull/20528

MisterMakerNL, 2025-04-11:
a bit weird that it shows the formula on the background.

furgo16, 2025-04-11:
@MisterMakerNL any pointers on where it could more or less be, so that it can be extracted from that PR for a quick fix?

MisterMakerNL, 2025-04-12:
I do think fixing this introduces another old but where you get double values displayed:

[image]

MisterMakerNL, 2025-04-12:
Tuned a bit 
[image]

furgo16, 2025-04-17:
@MisterMakerNL this looks good, thanks. Do you have any branch I can test? Alternatively, if you paste a snippet, happy to submit a hotfix PR, if that helps.

MisterMakerNL, 2025-04-17:
@furgo16 yea, here https://github.com/MisterMakerNL/FreeCAD_MisterMaker/tree/Stylesheet_upgrade
I could use some help with the preference menu if you are able to do so.
The issue is that the menu bar is hardcoded with an rgba which makes the background darker.
But since these are buttons I'd like to make it lighter not darker.

furgo16, 2025-04-17:
>  yea, here https://github.com/MisterMakerNL/FreeCAD_MisterMaker/tree/Stylesheet_upgrade

@MisterMakerNL thanks! I can try, but I might need some guidance (no experience with FreeCAD stylesheets). So I have some questions:

1. The changes we're talking about are in this commit: https://github.com/FreeCAD/FreeCAD/commit/bcf03986f724c9cb3198e3d1fdbeb82d54089fa2#diff-3d3fb90f686b31f298bb79fdd6708b1847001f9b1f9a0dbc9ee5d666ae7fb5ba, correct?
2. In that commit, are the changes to fix this issue (Property Editor) mixed with other changes related to the Preferences dialog? If so, which are which, and are they related?

furgo16, 2025-04-17:
@MisterMakerNL thanks, that part is clear, then. Still, on the `user_dark.qss` there are quite a lot of changes, some of which are not related to this issue (I think).

If I get this right, could you roughly point me to the changes I should extract to fix this issue for the Property Editor?

Syres916, 2025-04-17:
> The issue is that the menu bar is hardcoded with an rgba which makes the background darker. But since these are buttons I'd like to make it lighter not darker.

To use an extreme example:

```
QDialog#Gui__Dialog__DlgPreferences > QFrame {
    background-color: #ff0000; /* red backgroundarea */
}

QDialog#Gui__Dialog__DlgPreferences > QFrame QFrame {
    background-color: #0000ff; /* blue background area */
}
```

gives:

[image]

@MisterMakerNL is that what you were looking for?

MisterMakerNL, 2025-04-17:
@Syres916 That is awesome I tried so many of these things, non worked, but you figured it out!!
I do had it changed too:
```
QDialog#Gui__Dialog__DlgPreferences {
    background-color: @ThemeGradientColor1;
}

```

[image]

furgo16, 2025-04-19:
I've backported the changes in https://github.com/FreeCAD/FreeCAD/commit/bcf03986f724c9cb3198e3d1fdbeb82d54089fa2 to main as advised, but they don't seem to fix this issue.

Any advice would be welcome, thanks.

MisterMakerNL, 2025-04-19:
The hacks on top of the stylesheet seem to be causing this issue.
Not sure if you can remove them without any other issues.
I do have them removed from my new stylesheet, which I am finishing off.
Although I still have to do the light stylesheet and go over the overlay stylesheets.
[image]

furgo16, 2025-04-19:
At this point, I don't even know where to look at for a fix, so I give up.

I'll try FreeCAD Light again when https://github.com/FreeCAD/FreeCAD/pull/20528 is merged.

maxwxyz, 2025-05-26:
@alfrix is this also fixed in https://github.com/FreeCAD/FreeCAD/pull/21571

alfrix, 2025-05-26:
> [@alfrix](https://github.com/alfrix) is this also fixed in [#21571](https://github.com/FreeCAD/FreeCAD/pull/21571)

No, this is an issue with the stylesheets